earlephilhower/arduino-pico #1984: Problem with Signed Updates and using second core

Wxchris071, 2024-02-05:
The MCU is stuck when doing an Update with a signed binary and using setup1() and loop1() for another task.
Please see my example Code.
I think it gets stuck in Line 275 of Updater.cpp: ` _fp.close();`
[HTTPSUpdatesketch.zip](https://github.com/earlephilhower/arduino-pico/files/14165215/HTTPSUpdatesketch.zip)

earlephilhower, 2024-02-08:
I am unable to reproduce any hang with your sketch.  All I did was use the SSL certs from the HTTPS examples, update the IP configs, and use the pre-made signing key and infrastructure from the SignedOTA examples.  No hang or crash, and rebooted fine.

````
#define FWVer 0001
const char* update_username = "admin";
const char* update_password = "admin";

#include <W5100lwIP.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServerSecure.h>
#include <lwip/netif.h>
#include <HTTPUpdateServer.h>

const char ca_cert[]  = PROGMEM R"EOF(
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
)EOF";

// The server's private key which must be kept secret
const char server_private_key[] = PROGMEM R"EOF(
-----BEGIN RSA PRIVATE KEY-----
MIIEpQIBAAKCAQEA9UoHBtn4oNKXjRgIOQ/rLxK/iI0a8Q5mDxhfuwa9//FkftSI
IFY8UhGk2YNJpnfKOyYWqbqwuJhIZJ2sEIWp2301OnavuGBrpKOgBJJljgH2l/4Z
uUPLmbXQoPH3QEUpd08wItP6jvo3H0+kK/DXcDzjzAc0bdukGAi37D+8kJfduRGf
zIRgBToZ0xth7u/jjqFfeRjlSaB4kWv3HPWSroDNisy39QeE3M5L5ByQlpjQ4Za+
ZM2yljfEJtsW0gOQJ0FLwJkDD6K2eh++dZYpAReW3qi0+HmePDL47W39D5ns4wNh
BofTYhDHfXzz+RbcXM5jfaScRHW7OOZE76OEDwIDAQABAoIBAQDKov5NFbNFQNR8
djcM1O7Is6dRaqiwLeH4ZH1pZ3d9QnFwKanPdQ5eCj9yhfhJMrr5xEyCqT0nMn7T
yEIGYDXjontfsf8WxWkH2TjvrfWBrHOIOx4LJEvFzyLsYxiMmtZXvy6YByD+Dw2M
q2GH/24rRdI2klkozIOyazluTXU8yOsSGxHr/aOa9/sZISgLmaGOOuKI/3Zqjdhr
eHeSqoQFt3xXa8jw01YubQUDw/4cv9rk2ytTdAoQUimiKtgtjsggpP1LTq4xcuqN
d4jWhTcnorWpbD2cVLxrEbnSR3VuBCJEZv5axg5ZPxLEnlcId8vMtvTRb5nzzszn
geYUWDPhAoGBAPyKVNqqwQl44oIeiuRM2FYenMt4voVaz3ExJX2JysrG0jtCPv+Y
84R6Cv3nfITz3EZDWp5sW3OwoGr77lF7Tv9tD6BptEmgBeuca3SHIdhG2MR+tLyx
/tkIAarxQcTGsZaSqra3gXOJCMz9h2P5dxpdU+0yeMmOEnAqgQ8qtNBfAoGBAPim
RAtnrd0WSlCgqVGYFCvDh1kD5QTNbZc+1PcBHbVV45EmJ2fLXnlDeplIZJdYxmzu
DMOxZBYgfeLY9exje00eZJNSj/csjJQqiRftrbvYY7m5njX1kM5K8x4HlynQTDkg
rtKO0YZJxxmjRTbFGMegh1SLlFLRIMtehNhOgipRAoGBAPnEEpJGCS9GGLfaX0HW
YqwiEK8Il12q57mqgsq7ag7NPwWOymHesxHV5mMh/Dw+NyBi4xAGWRh9mtrUmeqK
iyICik773Gxo0RIqnPgd4jJWN3N3YWeynzulOIkJnSNx5BforOCTc3uCD2s2YB5X
jx1LKoNQxLeLRN8cmpIWicf/AoGBANjRSsZTKwV9WWIDJoHyxav/vPb+8WYFp8lZ
zaRxQbGM6nn4NiZI7OF62N3uhWB/1c7IqTK/bVHqFTuJCrCNcsgld3gLZ2QWYaMV
kCPgaj1BjHw4AmB0+EcajfKilcqtSroJ6MfMJ6IclVOizkjbByeTsE4lxDmPCDSt
/9MKanBxAoGAY9xo741Pn9WUxDyRplww606ccdNf/ksHWNc/Y2B5SPwxxSnIq8nO
j01SmsCUYVFAgZVOTiiycakjYLzxlc6p8BxSVqy6LlJqn95N8OXoQ+bkwUux/ekg
gz5JWYhbD6c38khSzJb0pNXCo3EuYAVa36kDM96k1BtWuhRS10Q1VXk=
-----END RSA PRIVATE KEY-----
)EOF";

// The server's public certificate which must be shared
const char server_cert[] = PROGMEM R"EOF(
-----BEGIN CERTIFICATE-----
MIIDSzCCAjMCCQD2ahcfZAwXxDANBgkqhkiG9w0BAQsFADCBiTELMAkGA1UEBhMC
VVMxEzARBgNVBAgMCkNhbGlmb3JuaWExFjAUBgNVBAcMDU9yYW5nZSBDb3VudHkx
EDAOBgNVBAoMB1ByaXZhZG8xGjAYBgNVBAMMEXNlcnZlci56bGFiZWwuY29tMR8w
HQYJKoZIhvcNAQkBFhBlYXJsZUB6bGFiZWwuY29tMB4XDTE4MDMwNjA1NDg0NFoX
DTE5MDMwNjA1NDg0NFowRTELMAkGA1UEBhMCQVUxEzARBgNVBAgMClNvbWUtU3Rh
dGUxITAfBgNVBAoMGEludGVybmV0IFdpZGdpdHMgUHR5IEx0ZDCCASIwDQYJKoZI
hvcNAQEBBQADggEPADCCAQoCggEBAPVKBwbZ+KDSl40YCDkP6y8Sv4iNGvEOZg8Y
X7sGvf/xZH7UiCBWPFIRpNmDSaZ3yjsmFqm6sLiYSGSdrBCFqdt9NTp2r7hga6Sj
oASSZY4B9pf+GblDy5m10KDx90BFKXdPMCLT+o76Nx9PpCvw13A848wHNG3bpBgI
t+w/vJCX3bkRn8yEYAU6GdMbYe7v446hX3kY5UmgeJFr9xz1kq6AzYrMt/UHhNzO
S+QckJaY0OGWvmTNspY3xCbbFtIDkCdBS8CZAw+itnofvnWWKQEXlt6otPh5njwy
+O1t/Q+Z7OMDYQaH02IQx3188/kW3FzOY32knER1uzjmRO+jhA8CAwEAATANBgkq
hkiG9w0BAQsFAAOCAQEAnDrROGRETB0woIcI1+acY1yRq4yAcH2/hdq2MoM+DCyM
E8CJaOznGR9ND0ImWpTZqomHOUkOBpvu7u315blQZcLbL1LfHJGRTCHVhvVrcyEb
fWTnRtAQdlirUm/obwXIitoz64VSbIVzcqqfg9C6ZREB9JbEX98/9Wp2gVY+31oC
JfUvYadSYxh3nblvA4OL+iEZiW8NE3hbW6WPXxvS7Euge0uWMPc4uEcnsE0ZVG3m
+TGimzSdeWDvGBRWZHXczC2zD4aoE5vrl+GD2i++c6yjL/otHfYyUpzUfbI2hMAA
5tAF1D5vAAwA8nfPysumlLsIjohJZo4lgnhB++AlOg==
-----END CERTIFICATE-----)EOF";

const char updatekey[] = PROGMEM R"EOF(
-----BEGIN PUBLIC KEY-----
...
-----END PUBLIC KEY-----
)EOF";

Wiznet5100lwIP eth0(17);
byte mac[] = { 0x64, 0x61, 0x69, 0x6E, 0x6F, 0x78 };
WebServerSecure server(443);
ServerSessions serverCache(5);
//#define CONFIG_IPV6_STATIC_ADDR "fe80::200:dcff:fe57:5761"
IPAddress ip(192,168,1,184);
IPAddress gw(192,168,1,1);
IPAddress netmask(255,255,255,0);
HTTPUpdateServerSecure updater(true);
BearSSL::PublicKey signPubKey(updatekey);
BearSSL::HashSHA256 hash;
BearSSL::SigningVerifier sign(&signPubKey);

void setup() {
    Serial.begin(115200);
    eth0.config(ip,gw,netmask);
    delay(100);
    //ip6_addr_t ip6ip = {0};
    //ip6addr_aton(CONFIG_IPV6_STATIC_ADDR, (ip6_addr_t*)&ip6ip.addr);
    delay(100);
    eth0.begin(mac);
    delay(100);
    //netif_add_ip6_address(eth0.getNetIf(),(ip6_addr_t*)&ip6ip.addr,0);
    delay(100);
    configTime(3 * 3600, 0, "192.168.1.1", "pool.ntp.org");
    server.getServer().setRSACert(new BearSSL::X509List(server_cert), new BearSSL::PrivateKey(server_private_key));
    server.getServer().setCache(&serverCache);
    server.begin();
    updater.setup(&server,"/update",update_username,update_password);
// Signing key in sketch dir, so no need to manually add it, plus the IDE will auto-sign any updated binaries w/o extra work
//    Update.installSignature(&hash, &sign);
}

void loop() {
    server.handleClient();
}

#if 1
void setup1() {
    delay(5000);
    Serial.println(FWVer);
}

void loop1() {
    delay(1);
}
#endif
````

FWIW, you may want to try moving to the IRQ based Ethernet just merged, it seems to make SSL much faster.  Only change for me was
````
Wiznet5100lwIP eth0(17, SPI, 21);
````

I'll need a failing sketch to look at this, because so far it all seems fine in my testing (about 7 uploads including the IRQ and polling based drivers).

Wxchris071, 2024-02-09:
Interesting..
I'm still having the same problem even with your Code.
Tried it with my custom PCB and an official W5100S-EVB-Pico with same result.
Next Step i'm going to try in Arduino IDE.

Wxchris071, 2024-02-09:
Problem solved.
Somehow it didn't like my Public Key. 
it works with the example key

closed 2024-02-09 by earlephilhower