earlephilhower/arduino-pico #1995: W5500 connected() not dynamically changing

maxgerhardt, 2024-02-09:
Consider the following code
```cpp
#include <W5500lwIP.h>

#define WIZNET_RSTn 20
#define WIZNET_CS   17
#define WIZNET_INT  21
Wiznet5500lwIP eth(WIZNET_CS /* chip select */, SPI, WIZNET_INT /* interrupt */);

void setup() {
  // Set up SPI pinout to match your HW
  SPI.setRX(16);
  SPI.setCS(17);
  SPI.setSCK(18);
  SPI.setTX(19);

  // get the chip out of reset
  pinMode(WIZNET_RSTn, OUTPUT);
  digitalWrite(WIZNET_RSTn, LOW);
  delayMicroseconds(500);
  digitalWrite(WIZNET_RSTn, HIGH);
  delay(1);

  Serial.begin(115200);
  delay(3000);
  Serial.println();
  Serial.println();
  Serial.println("Starting Ethernet port");

  // Start the Ethernet port
  if (!eth.begin()) {
    Serial.println("No wired Ethernet hardware detected. Check pinouts, wiring.");
    while (1) {
      delay(1000);
    }
  }
}

void loop() {
  Serial.println("Ethernet connected: " + String(eth.connected()));
  Serial.print("IP address: ");
  Serial.println(eth.localIP());
  delay(1000);
}
```

This uses the `Wiznet5500lwIP` with the interrupt pin. The pinout is the same as the [Wiznet W5500 EVB Pico](https://docs.wiznet.io/Product/iEthernet/W5500/w5500-evb-pico).

It is observed that, when the Ethernet cable is initially unplugged from the W5500, it prints
```
Ethernet connected: 0
IP address: (IP unset)
```
as expected. However, when the Ethernet cable is now inserted, this state does not change, although the green and orange LEDs on the Ethernet jack light up / blink.

In reverse, when the Pico is booted with the Ethernet cable already plugged in, it will output.
```
IP address: 192.168.0.69
Ethernet connected: 1
```
And now even when the Ethernet cable is unplugged, it does not change state. It will still report "connected" with the Ethernet cable removed.

This highly indicates that the interrupt-based state change of the Ethernet interface is not working properly.

maxgerhardt, 2024-02-09:
Addendum: This actually also fails when not giving the interrupt pin in the constructor (i.e. polling mode).

earlephilhower, 2024-02-09:
This is expected behavior.  Ethernet PHY status is via `isLinked()`, not `connected()` which is only looking at IP addressing.  It's done this way mostly for backward compatibility.  If you try:
````
void loop() {
  Serial.println("Ethernet connected: " + String(eth.connected()) + " isLinked:" + String(eth.isLinked()));
  Serial.print("IP address: ");
  Serial.println(eth.localIP());
  delay(1000);
}
````

You'll see something like
````
15:30:16.874 -> Starting Ethernet port
15:30:16.874 -> Ethernet connected: 0 isLinked:1
15:30:16.874 -> IP address: (IP unset)
15:30:17.866 -> Ethernet connected: 1 isLinked:1
15:30:17.866 -> IP address: 192.168.1.229
15:30:18.859 -> Ethernet connected: 1 isLinked:1
15:30:18.859 -> IP address: 192.168.1.229
15:30:19.885 -> Ethernet connected: 1 isLinked:1
15:30:19.885 -> IP address: 192.168.1.229
15:30:20.877 -> Ethernet connected: 1 isLinked:1
15:30:20.877 -> IP address: 192.168.1.229
15:30:21.871 -> Ethernet connected: 1 isLinked:0
15:30:21.871 -> IP address: 192.168.1.229
15:30:22.863 -> Ethernet connected: 1 isLinked:0
15:30:22.863 -> IP address: 192.168.1.229
15:30:23.889 -> Ethernet connected: 1 isLinked:0
15:30:23.889 -> IP address: 192.168.1.229
15:30:24.882 -> Ethernet connected: 1 isLinked:0
15:30:24.882 -> IP address: 192.168.1.229
15:30:25.875 -> Ethernet connected: 1 isLinked:0
15:30:25.875 -> IP address: 192.168.1.229
15:30:26.867 -> Ethernet connected: 1 isLinked:0
15:30:26.867 -> IP address: 192.168.1.229
15:30:27.860 -> Ethernet connected: 1 isLinked:1
15:30:27.893 -> IP address: 192.168.1.229
15:30:28.886 -> Ethernet connected: 1 isLinked:1
15:30:28.886 -> IP address: 192.168.1.229
````

It may be worthwhile to `&&` `isLinked` into the `connected()` logic, though...thoughts?

earlephilhower, 2024-02-09:
Oh, and the W5100 does not report PHY state, or at least the driver here doesn't know how to get that bit of info.

maxgerhardt, 2024-02-09:
Hmm then it was a API misunderstanding of mine. The `connected()` and `isLinked` functions *are* properly documented, in my IDE's mousehover.

closed 2024-02-10 by earlephilhower