earlephilhower/arduino-pico #1997: FreeRTOS examples (Multicore_FreeRTOS.ino and StaticMulticore-FreeRTOS.ino) STILL did not work.

mohdrais, 2024-02-11:
I have replaced the old variantHooks.cpp with the newest version (#1992) but STILL not working.

earlephilhower, 2024-02-11:
The examples run fine with `git master head`.  I suggest not manually copying files but instead following the `git` checkout method described in the docs.

Output from the FreeRTOS-Multicore demo this morning:
````
C1: Stay on target...
09:47:45.117 -> C0: Blue leader standing by...
09:47:45.117 -> # Tasks: 9
09:47:45.117 -> ID, NAME, STATE, PRIO, CYCLES
09:47:45.117 -> 0: CORE0            Running    4 25403610
09:47:45.117 -> 1: IDLE1            Running    0 323825611
09:47:45.117 -> 2: IDLE0            Ready      0 490590608
09:47:45.117 -> 3: BLINK            Blocked    1 579540
09:47:45.117 -> 4: CORE1            Blocked    4 46210091
09:47:45.117 -> 5: USB              Blocked    6 375677499
09:47:45.117 -> 6: IdleCore1        Suspended  7 17092
09:47:45.117 -> 7: IdleCore0        Suspended  7 44201000
09:47:45.117 -> 8: Tmr Svc          Blocked    2 20666
09:47:45.117 -> val: 65
09:47:45.944 -> C1: Stay on target...
09:47:46.115 -> C0: Blue leader standing by...
09:47:46.115 -> # Tasks: 9
09:47:46.115 -> ID, NAME, STATE, PRIO, CYCLES
09:47:46.115 -> 0: CORE0            Running    4 25795088
09:47:46.115 -> 1: IDLE1            Running    0 452708774
09:47:46.115 -> 2: IDLE0            Ready      0 622680791
09:47:46.115 -> 3: BLINK            Blocked    1 587653
09:47:46.115 -> 4: CORE1            Blocked    4 46236160
09:47:46.115 -> 5: USB              Blocked    6 381076493
09:47:46.115 -> 6: IdleCore0        Suspended  7 44201000
09:47:46.115 -> 7: Tmr Svc          Blocked    2 20666
09:47:46.115 -> 8: IdleCore1        Suspended  7 17092
09:47:46.115 -> val: 66
09:47:46.936 -> C1: Stay on target...
````

closed 2024-02-11 by earlephilhower

mohdrais, 2024-02-13:
Yes it works on Pico but not on *Pico W. *How to make it work on Pico W?

earlephilhower, 2024-02-13:
#2004

kevnikos, 2024-02-13:
> #2004

Timely, this mean a fix is coming?  I was just trying to get FreeRTOS working on my Pico W and was scratching my head.

mohdrais, 2024-02-14:
Thank you so much. Now they work for both Pico and Pico W.