earlephilhower/arduino-pico #2003: Update the core to support debugging with Arduino IDE 2.3 and Arduino CLI 0.35.0

kittaakos, 2024-02-13:
Please make the platform's debugging feature compatible with the latest Arduino IDE + CLI versions.

Arduino CLI's (`>=0.35.0`) debug feature is now based on a standard framework documented in the new [specifications and guidelines](https://arduino.github.io/arduino-cli/latest/platform-specification/#sketch-debugging-configuration). More details are in the [blog post](https://blog.arduino.cc/2024/02/07/ide-2-3-is-out-and-youll-love-the-new-debugging-features-in-it/).

Arduino IDE 2.3.0 uses the latest CLI APIs for the debugger feature. Starting from https://github.com/arduino/arduino-cli/pull/2391, the `programmer` is mandatory to retrieve the debug configuration from the CLI and start the debugger in IDE2. Otherwise, the debug feature will be disabled for the board in IDE2 (https://github.com/arduino/arduino-ide/issues/2368).

Please define the programmers so that the platform's debug capability can be used in Arduino IDE 2.3.0. Thank you!

Arduino IDE users using the current version of `Raspberry Pi RP2040 Boards` platform **must** use version `2.2.1` to enable the debugging.

----

Dev notes:

 - There is a new directive in the Arduino platform spec: [`programmer.default`](https://arduino.github.io/arduino-cli/dev/platform-specification/#set-a-default-programmer-for-a-board-since-arduino-cli-0350-arduino-ide-230)
 - Related for the ESP32 platform: https://forum.arduino.cc/t/ide-2-3-0-debugging-esp32s3-using-built-in-usb-jtag/1222051/2
 - Enable the debugging when the latest standards in the ESP32 platform: https://github.com/espressif/arduino-esp32/pull/9116

earlephilhower, 2024-02-13:
Thanks for the detailed info!  Would you like to take a stab at this?  You seem to have a good idea of the changes needed, already.  OTW, it'll go in the queue.

kittaakos, 2024-02-19:
The Arduino CLI will relax the constraints and make the `programmer` optional for the `debug --info` command: https://github.com/arduino/arduino-cli/pull/2540.

See confirmation: https://github.com/arduino/arduino-ide/issues/2368#issuecomment-1952364204.

I am closing this enhancement because it is not required now.

closed 2024-02-19 by kittaakos