earlephilhower/arduino-pico #2010: Wire.setSCL/Wire.setSDA halt the program.

hugo-sna, 2024-02-17:
Hello, 

I'm trying to build something with a Raspberry Pi Pico W and a PN532 but I can't get the I2C interfacing to work, it seems like Wire.setSDA and especially Wire.setSCL halts the program.
I'm not sure if it's my mistake or not but I don't really see what I did wrong.

[Here's my code](https://gist.github.com/hugo-sna/876d798d9e5c50e4dd48b34d2d27dee7) if anyone want to have a look.

LinusHeu, 2024-02-17:
Hi,
GPIO 21 can not be used for SDA. And GPIO 22 can not be used for SCL. This is a hardware limitation of how the things are wired internally inside the RP2040. 
Check out the pinout diagram here https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html or in the datasheet of the RP2040 to see what pins can be used for what functions.
```
#define NFC_SCL 22
#define NFC_SDA 21
```

hugo-sna, 2024-02-18:
Thanks a lot for helping me.