earlephilhower/arduino-pico #2011: Arduino IDE v2.3.1 for RP2040 has lost I2C capability

gdavids57, 2024-02-17:
Upon installing the Arduino IDE 2.3.1 release, my Adafruit Feather RP2040 RFM69 lost its I2C capabilities.  However, CircuitPython still communicates with my I2C peripherals as expected.  I observed the SDA/SCK lines with my scope and found no activity when using Arduino but observed correct activity with CircuitPython.  I suspect something has changed in the RP2040 Wire implementation as the IDE works correctly with my Arduino Nano 33 IOT board.  I have "package_rp2040_index.json" installed as an additional board and the IDE successfully connects with my Adafruit board.

earlephilhower, 2024-02-17:
I think this is just a mistaken board selection.

The last change to Wire was 3 months ago, per the GH web interface.  The Adafruit variants also are stable with the last update to them 7 months ago.

The IDE itself doesn't matter for generated code, only the core package version (3.7.2 as of y'day).  Also, the board you select will matter since they all chose different pinouts.  Please double check the board is the right one as the IDE may have chosen or changed your selections when you updated.  You can also use the `setSDA/setSCL` calls to specify GP pinouts directly (before calling `Wire.begin` of course, and only on legal pins).

If that doesn't work, you're going to need to provide a small MCVE with no external HW dependencies to reproduce.

gdavids57, 2024-02-18:
Thank you for the quick response.  I can’t determine any change to the board I selected but all of a sudden, my Adafruit Feather resumed I2C communications in the Arduino environment.

> On Feb 17, 2024, at 11:53 AM, Earle F. Philhower, III ***@***.***> wrote:
> 
> MCVE

closed 2024-02-18 by earlephilhower