earlephilhower/arduino-pico #2013: Picow compilation errors in lwIP

jhmaloney, 2024-02-19:
I'm getting some compilations errors when compiling for picow:

   22 | #include "CEspControl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico-w/lib1b0/lwIP_ESPHost/ESPHost.cpp.o] Error 1
In file included from /Users/johnmaloney/.platformio/packages/framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.cpp:21:
/Users/johnmaloney/.platformio/packages/framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.h:24:10: fatal error: CCtrlWrapper.h: No such file or directory

   24 | #include "CCtrlWrapper.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.

I use platformio. I started getting these errors after updating my arduino-pico installation yesterday.

earlephilhower, 2024-02-19:
As part of CI we build all the PicoW WiFi examples using `arduino-cli` so this is P.IO related.  Can you post your INI files and a small sketch showing the failure?

If you don't actually define `ESPHOSTSPI` the core will `ifdef` out all refs to the `ESPHost` library (not included, you must separately pull it in per the README in the lwip_ESPHost dir).

I don't use P.IO, but this reminds me of problems before where P.IO was being lazy and not actually probing to see what includes were *really* needed.  There's some sort of level setting option to have it actually evaluate `#define`s that fixes that problem.

jhmaloney, 2024-02-19:
Thanks for the super-quick response! This makes sense; I did not define ESPHOSTSPI so I was wondering why it was even trying to compile ESPHost.cpp.

I will try some other settings for the PlatformIO dependency detection system. Will let you know...

jhmaloney, 2024-02-19:
Got it! I just needed to add:

> lib_ignore = lwIP_ESPHost

To the PlatformIO entry for the pico-w. Thanks for putting me on the right track!

closed 2024-02-19 by earlephilhower