earlephilhower/arduino-pico #2023: Serial Monitor not functional under linux?

poetaster, 2024-02-25:
Running ubuntu 22 with IDE 1.8.1{0,9}  running 3.7.2., I get no output from the Serial monitor. All sketches upload fine and function as intended. Just no serial monitor. I can hack around it with projects that have an SSD1306 but that's not always possible :)

To compare, I ran some Circuitpython scripts:and 
```screen /dev/ttyACM0 115200```
 and immediately have a monitor and can break out, start stop etc.

Some posts on the pi forumns pointed to a possible issue with linux kernels but,  who knows?

``
uname -a
Linux w541 5.15.0-97-lowlatency #107-Ubuntu SMP PREEMPT Fri Feb 9 11:16:34 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
````

earlephilhower, 2024-02-25:
I only run Linux, as do many others, and the USB serial output is fine here.  Can you try `minicom` instead of `screen`?  Possibly `screen` does not set the virtual handshake signals properly and the chip doesn't get acknowledgment that it's safe to send data.  Some Windows terms don't do that as well.

If `minicom` runs then look at https://github.com/earlephilhower/arduino-pico/blob/master/docs/serial.rst#void-serialignoreflowcontrolbool-ignore and add it to your app and try again.

poetaster, 2024-02-25:
> I only run Linux, as do many others, and the USB serial output is fine here. Can you try `minicom` instead of `screen`? Possibly `screen` does not set the virtual handshake signals properly and the [...]

Screen, as tmux, are standard tools of the trade, but I had also tried minicom  which also shows nothing (on the pico, fine on the arduinos) .  With a most basic example (fade led), ALL of these will work. I just tried that to try to narrow down when this happens. So it just seems that all the sketches I'm working on are complex enough to cause some issue.

> If `minicom` runs then look at https://github.com/earlephilhower/arduino-pico/blob/master/docs/serial.rst#void-serialignoreflowcontrolbool-ignore and add it to your app and try again.

So, under very limited circumstances, minicom and screen  work. I'm guessing that it has something to do with I2C and/or using TX/RX (midi/clock) Both https://github.com/poetaster/scarp/tree/main/firmware/Pikobeats and the simple waveform example from the MCP4275 driver (which work fine otherwise) fail to work with Serial. I'm not sure what it is you're asking me to include? A function implemention? I'm a bit tired so I'm probably just not grokking . 

Thanks!

poetaster, 2024-02-25:
I did try:

```
void setup()
{
  Serial.begin(115200);
  Serial.ignoreFlowControl(true);
```

which makes no difference.

poetaster, 2024-02-25:
And, just for good measure, this is the example from the MCP4725 driver:
https://github.com/RobTillaart/MCP4725/blob/master/examples/MCP4725_wave_generator_RP2040/MCP4725_wave_generator_RP2040.ino

poetaster, 2024-02-25:
Ok, I just took apart the more simple example of the MCP driver and structured it so that the loop will print regardless of error conditions and it does produce output (also in the arduino Serial monitor). So, I'm guessing it must be an error condition. Sadly, the same code works without fail on the arduino (nano, last test). Must be somethink pico sdk specific. 

I'll try to to get my functional audio sketch (which works well enough to time external equipment while outputting audio and updating a display) outputting Serial by putting in selective outputs. In the MCP case, the error condition appeared in setup (no serial output), but serial output was fine in the main loop. Does that suggest anything?

In any case,  looks like I'll have debug more on my end. Thanks!

earlephilhower, 2024-02-25:
Good luck!

closed 2024-02-25 by earlephilhower