earlephilhower/arduino-pico #2027: cstdint not working correctly on platformio

ricardomatias, 2024-02-27:
I get a load of errors like the one below when building. Strangely it worked fine last week.

```ini
[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino

[env:adafruit_feather]
board = adafruit_feather
board_build.core = earlephilhower
board_build.usb_product = Melseq
upload_protocol = picotool
lib_deps = 
	playa=symlink:///Users/phoenix/Projects/cpp/audio/playa-cpp
	adafruit/Adafruit SH110X@^2.1.10
	adafruit/Adafruit GFX Library@^1.11.9
	adafruit/Adafruit TinyUSB Library@^3.0.0
	adafruit/Adafruit NeoPixel@^1.12.0
build_flags = 
	-Isonovolt/rp2040
	-D PICO_STDIO_USB
	-D PLAYA_PPQ=480
	-D USE_TINYUSB
```

```cpp
toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/12.3.0/cstdint:47:11: error: 'int8_t' has not been declared in '::'
   47 |   using ::int8_t;
      |           ^~~~~~
```

earlephilhower, 2024-02-27:
Please post a small MCVE that shows the issue.  P.IO is part of CI and passing so it's not reproducible w/o your code.

earlephilhower, 2024-03-05:
One week, no update or MCVE.  Closing.

closed 2024-03-05 by earlephilhower