earlephilhower/arduino-pico #2029: LittleFS fails to mount on W5500-EVB-Pico

MrZander, 2024-02-27:
Hello,
I am using the Wiznet W5500-EVB-Pico and I am not able to get LittleFS to mount.

I am running the SpeedTest example, the output is:
```
Beginning test
Unable to format(), aborting
done
```

If I am understanding the docs correctly, I should just be able to call `begin` and it will format and mount the FS automatically, correct?
This example fails:
```
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  delay(5000);
  if(!LittleFS.being())
  {
    Serial.println("Couldn't mount LittleFS");
  }
}

void loop() {
  delay(10000);
}
```
`Couldn't mount LittleFS`

I'm not really sure where else to look, any suggestions would be appreciated.

earlephilhower, 2024-02-27:
Make sure you've set a filesystem size.  By default all flash is for the app.  Look at Tools->Flash Size.

MrZander, 2024-02-27:
Yep, that was it.  
Thanks so much for the quick help & for this project!

closed 2024-02-27 by MrZander