earlephilhower/arduino-pico #2032: WiFiClient.write blocks intermittently

obdevel, 2024-03-02:
I have a proof-of-concept application for load testing the Pico W's network performance and stability.

Intermittently, WiFiClient.write will block for 5 seconds before returning 0. I see references to timeouts in the source but this doesn't seem to be available via the Arduino API. I guess it's a shortage of buffers in the underlying IP stack or the code borrowed from ESP8266, but 5 seconds seems abnormal.

I have three clients sending data to one server. Each message received is then forwarded to the other two clients.  Message size is around 20 bytes. My write function makes five attempts to send the data before giving up and closing the client socket.

I'm testing with 50 messages/sec from each client, so 150 messages/sec in and out, but I'd like to go higher.

e.g. debug output from my application, blocking time in microseconds. The wanted value implies a partial write.

```
    381848: > ** short write to client = 1, wrote = 0, wanted = 6, attempt = 1, took = 5003636
    386850: > ** short write to client = 1, wrote = 0, wanted = 6, attempt = 2, took = 5001356
    391853: > ** short write to client = 1, wrote = 0, wanted = 6, attempt = 3, took = 5002194
    396857: > ** short write to client = 1, wrote = 0, wanted = 6, attempt = 4, took = 5003340
    401859: > ** short write to client = 1, wrote = 0, wanted = 6, attempt = 5, took = 5001033
```

earlephilhower, 2024-03-03:
5 seconds is the internal timeout, so something somewhere is timing out.  Unfortunately without an MCVE there's not much anyone but yourself can do.

There is a limited packet space present in the LWIP implementation (it needs to preallocate things since it runs in IRQ-level).  You could try doubling it from 16k to 32k in [lwipopts.h](https://github.com/earlephilhower/arduino-pico/blob/master/tools/libpico/lwipopts.h) and [rebuilding](https://github.com/earlephilhower/arduino-pico/blob/master/tools/libpico/make-libpico.sh) `libpico.a` and re-running your full app.  If you're unable to rebuild and would like me to get you a test version of libpico.a, let me know.

obdevel, 2024-03-03:
Thanks. I have discovered the availableForWrite() method which should allow me to preempt this and back off. Initial testing shows it works mostly, with only one instance so far where it reported space but the write still failed.

At this stage, I'm trying to find the limits of capacity and throughput so I can set reasonable expectations. I may take up your kind offer but I also need to consider other people in my group who may want to use this code. Requiring them to patch the core would not be ideal.

I will also try to create an MCVE but even in reduced form it would be fairly large. And it would require multiple Picos, although for load testing I have a program running on my Mac.

obdevel, 2024-03-06:
I have been experimenting with UDP broadcasts as an alternative to writing the same data to multiple client TCP sockets. It works well and gets around the buffer starvation problem.

I have learned that the iPhone personal hotspot doesn't support UDP broadcasts ... sigh. All useful knowledge.

I'll close this issue and reopen if required. Thanks again.

closed 2024-03-06 by obdevel