earlephilhower/arduino-pico #2047: SerialUART.cpp: Hardware Flow Control configuration (RTS/CTS)

sam68ud, 2024-03-08:
Hi,
I believe there could be an error in the library regarding the uart hw flow control setting function:  SerialUART.begin()
In detail this function call: 
    _**uart_set_hw_flow(_uart, _rts != UART_PIN_NOT_DEFINED, _cts != UART_PIN_NOT_DEFINED);**_ 

The function is defined in uart.h: 
_**static inline void uart_set_hw_flow(uart_inst_t *uart, bool cts, bool rts)**_

As we can see the parameters rts and cts are exchanged with each other: the first parameter must be cts, second parameter rts.
There's no problem if you enable both rts and cts at the same time, but if you would like to use cts or rts flow control separately the behavior will be incorrect.

Regards

[image]

earlephilhower, 2024-03-08:
Looks like you spotted a bug, want to submit a PR to swap rts/cts in the call?

sam68ud, 2024-03-08:
Ok. Well.
Thanks

closed 2024-03-10 by earlephilhower

sam68ud, 2024-03-10:
> as completed via #2052

Thanks Earle.
Regards
Samuel

PR #2052 (2024-03-10): Unswap CTS/RTS enable on SeriaUART
Fixes #2047
commit a8d10671 (2024-03-10): Unswap CTS/RTS enable on SeriaUART (#2052)

Fixes #2047