earlephilhower/arduino-pico #2056: Waveshare RP2040-Tiny i2c demo crashes core 1

MNS26, 2024-03-18:
I,m currently working on a project where i need a few pico's are i2c slaves, i just ran into a issue where the moment i call Wire.requestFrom() it breaks the i2c slave.

i also tried it with the `TalkingToMyself` example and that has the exact same issue

I use VScode with platformIO
my build target is the `Waveshare RP2040-Zero` (the Tiny board is basically the same but w/o the extra pads underneath and with a external usb+reset board)

my current workaround is to have all the code in the receive call and when reading is needed just write it to the bus after receiving a specific packet.

earlephilhower, 2024-03-18:
What is the exact board you're using?  There is no Waveshare RP2040 Tiny in the boards list.

Are you using the exact, unmodified TalkingToMyself example?  I just ran it on a Pico with the 2 jumpers and it's working fine here.  If not, post the exact `setup()` code.

If you've changed the pinouts, then double check the pins are legal.  The core will `panic` if you ask it to use an illegal pin for a HW function.

MNS26, 2024-03-19:
Thanks for such a quick response!

the board i use is this one [Waveshare RP2040-Tiny Development Board](https://www.waveshare.com/rp2040-tiny.htm)
the one i'm technically building for is this one [RP2040-Zero](https://www.waveshare.com/product/rp2040-zero.htm)
as i said earlier the only difference is that this one has no pads on the back for the other GPIO pins and has a external usb board

this is its pinout of the RP2040-Tiny
[image]
[image]

this is the code from the example included (only added `#include <Arduino.h>` `void recv(int); `void req();` so it doesnt throw compile errors

```c++
// Simple I2C master and slave demo - Earle F. Philhower, III
// Released into the public domain
//
// Using both onboard I2C interfaces, have one master and one slave
// and send data both ways between them
//
// To run, connect GPIO0 to GPIO2, GPIO1 to GPIO3 on a single Pico
#include <Arduino.h>
#include <Wire.h>

void recv(int);
void req();

void setup() {
  Serial.begin(115200);
  delay(5000);
  Wire.setSDA(0);
  Wire.setSCL(1);
  Wire.begin();
  Wire1.setSDA(2);
  Wire1.setSCL(3);
  Wire1.begin(0x30);
  Wire1.onReceive(recv);
  Wire1.onRequest(req);
}

static char buff[100];
void loop() {
  static int p;
  char b[90];

  // Write a value over I2C to the slave
  Serial.println("Sending...");
  Wire.beginTransmission(0x30);
  sprintf(b, "pass %d", p++);
  Wire.write(b, strlen(b));
  Wire.endTransmission();

  // Ensure the slave processing is done and print it out
  delay(1000);
  Serial.printf("buff: '%s'\r\n", buff);

  // Read from the slave and print out
  Wire.requestFrom(0x30, 6);
  Serial.print("\nrecv: '");
  while (Wire.available()) {
    Serial.print((char)Wire.read());
  }
  Serial.println("'");
  delay(1000);
}

// These are called in an **INTERRUPT CONTEXT** which means NO serial port
// access (i.e. Serial.print is illegal) and no memory allocations, etc.

// Called when the I2C slave gets written to
void recv(int len) {
  int i;
  // Just stuff the sent bytes into a global the main routine can pick up and use
  for (i = 0; i < len; i++) {
    buff[i] = Wire1.read();
  }
  buff[i] = 0;
}

// Called when the I2C slave is read from
void req() {

}
```

this is the verbose build output
https://pastebin.com/QnUsi5EQ

earlephilhower, 2024-03-19:
Errr, your example code doesn't have anything running on the 2nd core (no `loop1`) and never replies anything because the req() callback is empty.  So I haven't really got any idea what you're trying to do here.

Please run the plain TalkingToMyself with the jumper from GP0->GP2 and GP1->GP3 (note these may not be pins 0-3 on the PCB but they should be labeled)....

earlephilhower, 2024-03-19:
There's nothing in the `req()` function you listed, so there will never be any data sent back with what you posted.  So, somehow you seem to have a modified example code.

If you check the original code it's setting data to return to the other I2C:  https://github.com/earlephilhower/arduino-pico/blob/175cbcdcdf6bac39854486315fd514c82c610215/libraries/Wire/examples/TalkingToMyself/TalkingToMyself.ino#L64-70

Also, since it's not running anything on core1 I'm still at a loss why you believe `Wire` crashes the other core?

MNS26, 2024-03-19:
oh interesting, i will try that one.

Is it possible that PlatformIO has a outdated version of the pico core?

MNS26, 2024-03-19:
hmm ok just tried that one and it works. Well that means my code breaks somewhere... (and got no idea where )

i wonder why the platformIO one is missing part of the code though

MNS26, 2024-03-19:
the weird part is that when i have nothing in yhe req in my code the netire slave is gone form the i2c bus

earlephilhower, 2024-03-19:
There is no official P.IO support for this or any other RP2040 stuff, AIUI, so you should be using the `git` or release files as described [here](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst).

In any case, looks like were' good here so closing.

closed 2024-03-19 by earlephilhower

MNS26, 2024-03-19:
The stip of WS2812 are controlled by Core 0 (the same core running I2C)
the internal WS2812 is controlled by Core 1

MNS26, 2024-03-19:
https://youtu.be/AmeiO_LWxh4
Core 0 runs i2c and handles the strip
Core 1 runs the internal one

MNS26, 2024-03-19:
> There is no official P.IO support for this or any other RP2040 stuff, AIUI, **so you should be using the `git` or release files as described [here](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst)**.
> 
> In any case, looks like were' good here so closing.

i personally would not consider this closed (yet)
i also checked and yes im running the one from git.

after i said it was working i also said it still dissapears from the bus (yes its not the EXACT demo)
i tried it while i commented the `Wire.onRequest()` out and it STILL crashes the chip (!)
i also tried reworking the example code you liked but still had it crash

this is the entire project in its current state  (i2c breaks)
there are 2 build options one for the USB HID part and one for the i2c slave (node)

(its a 7z since it was the smallest one but file extention set as zip to upload it)
[RP2040 HOTAS.7z.zip](https://github.com/earlephilhower/arduino-pico/files/14657901/RP2040.HOTAS.7z.zip)