earlephilhower/arduino-pico #2057: MIDI over USB using Adafruits TinyUSB causes lockups / crashes

CaptainCredible, 2024-03-18:
Ran into problems using midi over USB and the Adafruit TinyUSB stack. The device will intermittendly lock up if data is transmitted, but curiously not necessarily when the data is transmitted. 

I reproduced the problem with a stripped down version of the example sketch from the Adafruit TinyUSB library (MIDI library v5.0.2 and Arduino Pico v3.7.2):

`/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This sketch is enumerated as USB MIDI device. 
 * Following library is required
 * - MIDI Library by Forty Seven Effects
 *   https://github.com/FortySevenEffects/arduino_midi_library
 */

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;

MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup()
{
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif
  pinMode(LED_BUILTIN, OUTPUT);
  //usb_midi.setStringDescriptor("TinyUSB MIDI"); //causes device not to show up
  MIDI.begin(MIDI_CHANNEL_OMNI);
  MIDI.turnThruOff();
  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleNoteOff(handleNoteOff);
  Serial.begin(115200);
  // wait until device mounted
  while( !TinyUSBDevice.mounted() ) delay(1);
}

void loop()
{
// read any new MIDI messages
  MIDI.read();  
}

void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}
`

Under light loads (few midi messages) this sketch will often work fine, but sometimes lock up. Under heavy load (lots of midi messages) It will lock up a lot sooner. 

Not certain if this is an issue with the rp2040 core or with the Adafruit TinyUSB stack.

earlephilhower, 2024-03-18:
Sorry, but because this is HW related (I don't have any MIDI equipment or adapters for the RP2040) there's not much anyone but you can do to debug this.

I don't have experience with the Adafruit MIDI (just a wrapper for TinyUSB MIDI AIUI), but right off the bat I believe you're going to get callbacks from within a USB IRQ.  Which means that writing to `Serial`, which is also USB, **is not legal or safe** (Tiny
USB is NOT re-entrant!).  Can you dump to a UART port (`Serial1` instead, for example) instead?

My recommendation would be to hook up a PicoProbe and check what the Pico is doing when it seems to hang.  The core stack trace from `gdb`'s `where` is often enough to explain what's going on.  If not, you could dump it and post it here for others to look at.

CaptainCredible, 2024-03-19:
PicoProbe! Thats what I'll do then. I have been working in the Arduino IDE until now and as far as I can tell I need to be using something else like VSCode + platformio to use the debugging properly. 
P.S. the above example doesn't rely on any other hardware than a raspberry pi pico and a usb cable. MIDI can be sent and received by a computer using a plethora of free software like,MIDIOX(win) reaper, or even browser based alternatives like this: https://versioduo.com/webmidi-test/

earlephilhower, 2024-03-19:
> ...the above example doesn't rely on any other hardware than a raspberry pi pico and a usb cable...

Oops!  I was thinking of the old-style round DIN UART-based MIDI stuff.  USB's eaten that, too, I guess. :)

In any case, before trying to learn VSCode (I myself just run openocd and gdb in a command shell) remove that `Serial.xxxx` stuff from the note callbacks.  That's not legal if TinyUSB is calling your callback from a USB interrupt (which I am 99% sure it is) and the root of your crashes.  No need to run a debugger.

You should instead put note ons/note offs  into a IRQ-safe queue and have the main `loop` handle the printout or actual sound generation.

The Adafruit example may do it this way because for the basic Arduino boards `Serial` used to be a simple UART and not USB-based (AVR-type boards, for example).

CaptainCredible, 2024-03-24:
I tried removing al Serial.xxx and this did indeed remedy the problem. 

the code below blinks the LED to give visual feedback as to whether or not the pico has locked up and responds with MIDI messages to any incoming MIDI messages. I can send notes and midi clock to it at a rate of 999BPM (*24 clock ticks per beat) and the pico copes just fine and responds with MIDI messages in return.

`#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;

MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup()
{
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif
  pinMode(LED_BUILTIN, OUTPUT);
  MIDI.begin(MIDI_CHANNEL_OMNI);
  MIDI.turnThruOff();
  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleNoteOff(handleNoteOff);
  MIDI.setHandleClock(handleClock);
  while( !TinyUSBDevice.mounted() ) delay(1);
}

int incrementor = 0;
bool ledState = false;
void loop(){
// read any new MIDI messages
  MIDI.read();
  //while(MIDI.read()){
  //}
  incrementor++;
  incrementor = incrementor%10000;
  if(incrementor == 0){
    ledState = !ledState;
    if(ledState){
      digitalWrite(LED_BUILTIN, HIGH);
    } else {
      digitalWrite(LED_BUILTIN, LOW);
    }
  }
}

void handleClock(){
  int nombre = random(30,80);
  MIDI.sendNoteOn(nombre, 100, 5);
  MIDI.sendNoteOff(nombre, 100, 5);
}

void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  MIDI.sendNoteOn(pitch, velocity, channel);
}

void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  MIDI.sendNoteOff(pitch, velocity, channel);
}
`

closed 2024-03-24 by CaptainCredible