earlephilhower/arduino-pico #2068: Using ADC pins causes crash

aberoev, 2024-03-22:
Hello and thank you for the core!
 
I am trying to utilize all Pico gpios for encoders with a midi controller for my home studio, and it goes very well up until pins 26, 27, 28 (ADC pins)

When changing pins to 26, 27 the Pico crashes. All other pins work fine

Any help appreciated

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc {
  {0, 1},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};

void setup() {
  RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
pinMode(0, INPUT_PULLUP);
pinMode(1, INPUT_PULLUP);
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

earlephilhower, 2024-03-22:
This isn't an MCVE and doesn't reference any ADC pins.  Please make a small, self contained MCVE without needing special HW (or libraries) to demonstrate the problem or there's nothing anyone can do to help here.

If you can't isolate into a simple sketch others can run, then I suggest using PicoProbe to break into the core and see where things are when they appear hung.  FWIW, the core itself (nor the Pico RP2040 chip) doesn't really care about ADC/non-ADC pins.  Inputs are inputs and whether you use them thru the GPIO pad or the ADC input mux is up to you.

aberoev, 2024-03-22:
Thank you for getting back. I did a basic test, and the mentioned ADC pins sure work for both digital read and write

Somehow instantiating an encoder on these pins will not work for me. Using control-surface.h various encoder objects will result in a crash with the onboard gp25 led flashing. I tried [another encoder library](https://github.com/gbr1/rp2040-encoder-library) which works with every pin set, no crashing with the ADC pins, however they simply do not work. I wish I had the knowledge to troubleshoot further on my own, but Picoprobe is way over my head.

closed 2024-03-22 by earlephilhower