earlephilhower/arduino-pico #2071: RP2040 and QNEthernet library issue

5b4wn, 2024-03-24:
Dear @earlephilhower 

MANY THANKS FOR YOUR CORE libraries. Amazing work. I am trying to compile with the help of @ssilverman QNEthernet library which is an amazing ethernet library but running issues with some of the included files WString.h IPAddress.h  (https://github.com/ssilverman/QNEthernet/issues/70#issuecomment-2016607802).

Are you able to make some suggestions please?

BW
@5b4wn

JAndrassy, 2024-03-24:
why would you want to use QNEthernet with the Pico Core? Pico Core has its own LwIP base library which supports W5500

earlephilhower, 2024-03-24:
We already natively support LWIP on the WizNet and other Ethernet adapters.  See the LWIP_xxx examples.

We also hove those two headers as they're standard Arduino API.

closed 2024-03-24 by earlephilhower