earlephilhower/arduino-pico #2073: Please add library for enhanced dealing with EEPROM

woodlist, 2024-03-24:
It would be nice to get the easy way on EEPROM using like does the "Preferences" library on ESP32 platform.

earlephilhower, 2024-03-25:
There is no reason to use `EEPROM` for new code in 2024.  `LittleFS` and the `File` can do everything it can do, and about 100x more while being wear-leveled and power-fail safe.  Look at the examples in the `LittleFS` library or the [docs](https://github.com/earlephilhower/arduino-pico/blob/master/docs/fs.rst) for a quick introduction.

closed 2024-03-25 by earlephilhower

woodlist, 2024-03-25:
Thanks for the reply. However, I was not satisfied with your answer. I know what is the LittleFs and how to deal with it. But too many things needed to be written to get it worked.
Let we say, here is an instrument available to declare very simply-"Unforgettable int (or bool or float or etc) myUnforgetableIntegerArray[18]". That's all. The initialization could be done as in Preferences on ESP32, that is to say, if the variable data is empty, assign "something default".

woodlist, 2024-03-25:
Thanks for the reply. However, I was not satisfied with your answer. I know what is the LittleFs and how to deal with it. But too many things needed to be written to get it worked.
Let we say, here is an instrument available to declare very simply-"Unforgetable int (or bool or float or etc) myUnforgetableIntegerArray[18]". That's all. The initialization could be done as in Preferences on ESP32, that is to say, if the variable data is empty, assign "something default".

woodlist, 2024-03-25:
I came across Pico specification and it is found that the Pico has not EEPROM at all.