earlephilhower/arduino-pico #2082: Feature Request: Make wizphy_setphypmode accessable

traxanos, 2024-04-01:
Hello,

is it possible to make wizphy_setphypmode() accessible? I would like to switch from EthernetGeneric driver to lwip. But with some switches (e.g. cisco) there are problems with auto sense. But the main reason is that I want to limit the speed to 10mbits to save a lot of power.

greets traxanos

earlephilhower, 2024-04-07:
I think a generic Ethernet class method would be a good way of doing this.  This way it's integrated with the classes and we can implement the same logic, possibly, on other Ethernet adapters.

sstaub, 2024-08-13:
I have this possibility in my Ethernet library https://github.com/sstaub/Ethernet3
Maybe that parts of the code will help to integrate it to lwip.

earlephilhower, 2024-08-13:
Thanks!  It's simple enough to do but because all the Ethernet (wired and wireless) are through the same base class here I was just cogitating on how to properly expose it so you can set `100M` for wired or `BG` for wireless.  An extensible `enum` like you use in your own wrapper sounds like a good way to get there...