earlephilhower/arduino-pico #2085: Bluetooth Pairing doesn't persist across reprogramming

nWestie, 2024-04-02:
### Issue:
I'm looking at using SerialBT with the Pico W for a project, and can pair/connect just fine, receive messages, etc. However, whenever I re-upload code to the board, it makes me pair to it again. Pairing persists across powering the board off/on, but not reprogramming. Is this something that could be fixed? It seems like an issue of the flash where the pairing data is stored being wiped on program, and I'm not sure how to work around that. 

If it's useful, here's my test code:
```cpp
#include <SerialBT.h>

void setup() {
  // put your setup code here, to run once:
 SerialBT.begin();
 Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  if(SerialBT.available()){
    Serial.write(SerialBT.read());
  }
}

```

earlephilhower, 2024-04-07:
This is by design.  Bluetooth gives a callback at the IRQ level when it needs to store keys so we can't use something like a file. And there's no spot set away for BT because when I started this core there was no BT or WiFi.  Maybe moving it to the last sector before end of program space would be an option 

Right now, the pairing information is overwritten right inside the application's flash constant memory.  That is, it's actually part of the program.

nWestie, 2024-04-10:
Ok, thanks. I was hoping to use it to debug a mobile robot, so probably will just skip it for now.

closed 2024-04-10 by nWestie