earlephilhower/arduino-pico #2086: ARDUCAM PICO4ML and NinaWiFi

fethicandan, 2024-04-03:
Hi,

I am trying to communicate with pico4ml and NinaWifi Module. Actually I am using Arduino Nano Connect hardware and Pico4ml system. Beucase I wanted to use IMU and Camera simultaneously. Camera and IMU are working on RP2040 Connect using pico4ml, but I want to use WiFiNnina module as well. 

Do you have any suggestions ?

P.S. I tried to add HM01B0 camera library into Arduino connect library but it is not working. Moreover, I used WifiNina Generic library, the system didn't recognise the SPI WifiNina module.

JAndrassy, 2024-04-03:
you could try the esp-hosted firmware for NINA with the WiFi library in this core
https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_ESPHost (follow the link to ESPHost library. it has the fw built for Nano Connect)

fethicandan, 2024-04-03:
> you could try the esp-hosted firmware for NINA with the WiFi library in this core https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_ESPHost (follow the link to ESPHost library. it has the fw built for Nano Connect)

Which board may I choose? Raspberry Pi Pico or RP2040? 
Even if I choose one of them. How can I use the HM01B0 camera and ICM20498 IMU? Beucase of these libraries I selected pico4ml. Otherwise, I can use directly Arduino RP2040 Connect with NinaWifi, am I right?