earlephilhower/arduino-pico #2088: Enhancement request: add Pico LITE PURPLE 16MB board

aotta, 2024-04-06:
Is possible to add to supported boards the Lite purple 16mb "clone" board? they are cheap and offer more exposed pins and 16mb flash: https://forums.raspberrypi.com/viewtopic.php?t=337976

earlephilhower, 2024-04-06:
For something like that, your best option is to use the Generic board.  You can select the flash size and type from the menus to match the generic AX version.

closed 2024-04-06 by earlephilhower

aotta, 2024-04-06:
Thank you @earlephilhower , but just to share with other users some more info, i tested Generic board but it stops recognize the board after flashing if used overclock, while with others board configuration (i used "Raspberry pico" or "Waveshare plus 16mb") overclocks works fine.
But of course it's becouse something not documented about the "purple" and "black" clones (yes, i had same issues with the black ones).

earlephilhower, 2024-04-07:
You can select a flash with /4 (ie system clock divided by 4) from the flash menu to show down the interface.

What you're seeing seems to indicate they're using factory rejects or fakes.  Possibly fine for a hobbyist, but I'd never put them in the field for long term use.