earlephilhower/arduino-pico #2095: Unable to silence warnings from the framework/pico-sdk

AgainPsychoX, 2024-04-08:
While compiling my project there are a lot of warnings coming vendor code I can't directly change, so I wan to silence it to focus on my own project files. 

### My setup

I use PlatformIO include Visual Studio Code on Windows.

I use combination of `build_unflags`/`build_src_flags` to apply warnings only to my sourced compilation units, and then extra script to change paths in `CPPPATH` to use `-isystem` to ignore the warnings.

My `platformio.ini`:
```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
; ...
build_unflags = 
	-Wall -Wextra -Wpedantic -Wvla ; hide warnings from framework & libraries
	-std=gnu++14 -std=gnu++17
build_flags =
	-std=gnu++20
; ...
	; Explicitly add toolchain headers (list from 'arm-none-eabi-g++.exe -v') to fix VS Code IntelliSense
	; List from `arm-none-eabi-g++.exe -v`, section `#include <...> search starts here`.
	-isystem $PROJECT_CORE_DIR/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.3.1
	-isystem $PROJECT_CORE_DIR/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.3.1/arm-none-eabi
	-isystem $PROJECT_CORE_DIR/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.3.1/backward
	-isystem $PROJECT_CORE_DIR/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/9.3.1/include
	-isystem $PROJECT_CORE_DIR/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/9.3.1/include-fixed
	-isystem $PROJECT_CORE_DIR/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include
build_src_flags = 
	-Wall -Wextra -Wpedantic
extra_scripts = 
	scripts/convert_sysincludes.py ; hide warnings from framework & libraries
```

I use extra script `scripts/convert_sysincludes.py` to convert some includes to be included as `-isystem`:
```py
# Script to help hide warnings from framework & libraries
# From https://community.platformio.org/t/silence-warnings-for-dependencies-external-libraries/33387/8

from pathlib import Path

# PlatformIO injected stuff (I wish it was typed... https://github.com/platformio/platformio-docs/issues/342)
Import("env" ,"projenv") # type: ignore
env, projenv, DefaultEnvironment = env, projenv, DefaultEnvironment # type: ignore

platform = env.PioPlatform()
if env.GetProjectOption("board_build.core") == 'earlephilhower':
	FRAMEWORK_DIR = Path(platform.get_package_dir("framework-arduinopico"))
else:
	FRAMEWORK_DIR = Path(platform.get_package_dir("framework-arduino-mbed"))
framework_includes = list()
filtered_cpppath = list()

# Apply these changes to current working env, the project env and the global env
for e in (env, projenv, DefaultEnvironment()):
	for p in e["CPPPATH"]:
		# Is the current include path inside the framework directory?
		if FRAMEWORK_DIR in Path(p).parents:
			framework_includes.append(p)
		else:
			filtered_cpppath.append(p)
	e.Replace(CPPPATH=filtered_cpppath)
	e.Append(CCFLAGS=[("-isystem", p) for p in framework_includes])

```

### **Sadly, it's not enough**.

Example warnings when compiling:
```log
In file included from C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:12,
                 from src\sampling.cpp:2:
C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_gpio/include/hardware/gpio.h:881:33: warning: ISO C++ does not permit named variadic macros [-Wvariadic-macros]
  881 | #define CU_REGISTER_DEBUG_PINS(p...) \
      |                                 ^~~
In file included from C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico/types.h:12,
                 from C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico.h:24,
                 from C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:10:
C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h: In function 'uint32_t xip_alias_check_addr(const void*)':
C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico/assert.h:35:34: warning: ISO C++ forbids braced-groups within expressions [-Wpedantic]
   35 | #define valid_params_if(x, test) ({if (PARAM_ASSERTIONS_ENABLED(x)) assert(test);})
      |                                  ^
C:\Users\PsychoX\.platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:95:5: note: in expansion of macro 'valid_params_if'
   95 |     valid_params_if(ADDRESS_ALIAS, rc >= XIP_MAIN_BASE && rc < XIP_NOALLOC_BASE);
      |    
```

Using `pio run --verbose` I can inspect final compiler arguments:
```
arm-none-eabi-g++ -o .pio\build\pico\src\sampling.cpp.o -c -std=gnu++20 -fno-exceptions -fno-rtti -isystem C:\Users\PsychoX\.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.3.1 -isystem C:\Users\PsychoX\.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.3.1/arm-none-eabi -isystem C:\Users\PsychoX\.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.3.1/backward -isystem C:\Users\PsychoX\.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/9.3.1/include -isystem C:\Users\PsychoX\.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/9.3.1/include-fixed -isystem C:\Users\PsychoX\.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include -Os -Werror=return-type -Wno-psabi -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -iprefixC:\Users\PsychoX\.platformio\packages\framework-arduinopico @C:\Users\PsychoX\.platformio\packages\framework-arduinopico\lib\platform_inc.txt -Wall -Wextra -Wpedantic -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\cores\rp2040 -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\cores\rp2040\api\deprecated -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\cores\rp2040\api\deprecated-avr-comp -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\include -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\variants\rpipico -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\libraries\EEPROM\src -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\libraries\LittleFS\src -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\libraries\SPI\src -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\cores\rp2040 -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\cores\rp2040\api\deprecated -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\cores\rp2040\api\deprecated-avr-comp -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\include -isystem C:\Users\PsychoX\.platformio\packages\framework-arduinopico\variants\rpipico -DPLATFORMIO=60114 -DARDUINO_RASPBERRY_PI_PICO -DARDUINO_ARCH_RP2040 -DUSBD_MAX_POWER_MA=500 -DUSER_SETUP_LOADED=1 -DST7796_DRIVER=1 -DTFT_WIDTH=320 -DTFT_HEIGHT=480 -DTFT_MISO=4 -DTFT_MOSI=3 -DTFT_SCLK=2 -DTFT_CS=5 -DTFT_DC=7 -DTFT_RST=6 -DTOUCH_CS=8 -DLOAD_GLCD=1 -DLOAD_FONT2=1 -DLOAD_FONT4=1 -DLOAD_FONT6=1 -DLOAD_FONT7=1 -DLOAD_FONT8=1 -DLOAD_GFXFF=1 -DSMOOTH_FONT=1 -DSPI_FREQUENCY=80000000 -DARDUINO=10810 -DARDUINO_ARCH_RP2040 -DF_CPU=133000000L -DBOARD_NAME=\"pico\" -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" 
-DUSB_PRODUCT=\"Pico\" -DSERIALUSB_PID=0x000a -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DLWIP_IPV6=0 -DARDUINO_VARIANT=\"rpipico\" -Iinclude -Isrc -I.pio\libdeps\pico\CRC32\src -I.pio\libdeps\pico\TFT_eSPI src\sampling.cpp
```

And I can see `-iprefixC:\Users\PsychoX\.platformio\packages\framework-arduinopico @C:\Users\PsychoX\.platformio\packages\framework-arduinopico\lib\platform_inc.txt` which the extra script I use can do nothing about (or rather would be quite hard) to make it as system (so warnings ignored) include.

While researching issue I found #615:
https://github.com/earlephilhower/arduino-pico/blob/2a256f9c25021deee92a5eece6137dbe1438e0ec/tools/platformio-build.py#L185-L192

* Shouldn't it all be `-isystem` anyways?
* Maybe I am missing something? Is there maybe easier way to I disable the warnings from framework?

AgainPsychoX, 2024-04-08:
### Workaround using more extra scripting

I updated my extra script as workaround, adding all the file parsing, then prefixes unrolling and changing the framework includes to use `-isystem`. Gist: https://gist.github.com/AgainPsychoX/45d55d4d7004cb6f20f7920807f72981 .

I guess it's not perfect, as now the whole process has to create temporary files because of how long the shell command is with all those arguments. Here is `pio run --verbose 2>&1 > verbose.log` after changing one file: Gist: https://gist.github.com/AgainPsychoX/0c4b897d110dfdcdb7ce9c0971be4d76 .

### Workaround using other include flag

Meanwhile I learnt there is `-iwithprefixbefore` (acts like `-I`) and `-iwithprefix` (acts like `-idirafter`). The priority of includes is different, but it seems working if I change them in the `platform_inc.txt` file - which makes my script modification I just wrote useless, welp. https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html

**Do you think could it be updated in the repo?** I don't want to create pull request, since I am not sure if it changes/breaks anything due to the include order changes; would require some intuition to judge it and testing to make sure, I imagine.

AgainPsychoX, 2024-04-08:
Update: I just found out that using second workaround (the one with `-iwithprefixbefore`) doesn't work for me: VS Code `c_cpp_properties.json` is missing stuff if it's just `-iwithprefix`, but it could work if code in `platformio-build.py` would be changed too (also to `-iwithprefix`):

https://github.com/earlephilhower/arduino-pico/blob/2a256f9c25021deee92a5eece6137dbe1438e0ec/tools/platformio-build.py#L46-L52

For now I will be using my extra script, but I hope you consider if `-iwithprefix` could be used in the official version. I could make pull request if you want, but again: it would require knowledge/testing if changed include order affects anything.

earlephilhower, 2024-04-10:
Those headers are used in the core and built in the CI without warnings.  Are you using a non standard compiler or set up?

AgainPsychoX, 2024-04-13:
Versions are in the logs:
```
PACKAGES: 
 - framework-arduinopico @ 1.30700.0+sha.88ccf0c (git+https://github.com/earlephilhower/arduino-pico.git#88ccf0c256dc717ff932adc9e2d84485214bf2e5) 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0, https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-w64-mingw32.arm-none-eabi-d04e724.240125.zip)
```