earlephilhower/arduino-pico #2096: My code doesn't see rp2040.rebootToBootloader();

fapplin, 2024-04-09:
Hi,

I just upgraded your library to the latest and I'm using the Arduino IDE 2.3.2. I get an error on rp2040.rebootToBootloader(); - which use to work in a previous version of your library and in the IDE 1.8. When I type "rp2040." I see all of the methods in the new IDE (which is nice), but I don't see rebootToBootloader() as one of the methods. Any ideas?

Frank

fapplin, 2024-04-09:
Just an added comment - it compiles just fine with your new library under the Arduino IDE 1.8.19.

earlephilhower, 2024-04-10:
Can you clarify?  Is it the code doesn't compile or the IDE doesn't hilight it?

fapplin, 2024-04-10:
The code doesn't compile.
======================================================================= 
The price of greatness is responsibility. - Winston Churchill

earlephilhower, 2024-04-10:
I'm traveling this week, but think I know what's going on.

Double check you're using this core and not the official Arduino one.  The Arduino core has no such function, only this one.

The actual source code and compile steps are self contained in this repo, so what you're seeing seems to be only possible if after changing the IDE it isn't using this core.

earlephilhower, 2024-04-15:
I just tried compiling the simple sketch below with 2.3.2 IDE:
````
void setup() {
rp2040.rebootToBootloader();
}
void loop() {
}
````
and did not encounter any errors.  I imagine it really was a matter of ending up with the Arduino official core and not this one.  Good luck!

closed 2024-04-15 by earlephilhower