earlephilhower/arduino-pico #2116: FYI: Build on Windows

traxanos, 2024-04-20:
Hi,

your repo is no more buildable on windows.

```
From https://github.com/STMicroelectronics/stm32u5xx_hal_driver
 * branch            2e1d4cdb386e33391cb261dfff4fefa92e4aa35a -> FETCH_HEAD
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32u5xx_hal_driver': checked out '2e1d4cdb386e33391cb261dfff4fefa92e4aa35a'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32wbxx_hal_driver': checked out '2c5f06638be516c1b772f768456ba637f077bac8'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/ti': checked out '143ed6cc20a7615d042b03b21e070197d473e6e5'
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long
```

earlephilhower, 2024-04-20:
We build under Windows as part of CI for every PR, and you're having trouble in the `pico-sdk` submodule which hasn't been updated in almost a year.

Looks like you're doing a `git submodule update --init --recursive` which isn't needed and grabs submodules many levels deep in the SDK->TinyUSB path.  If you *really* want to do this, then see the [docs on how to get long paths working in Windows](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst#important-steps-for-windows-users-before-installing).  It'll also take something like 2GB disk space for the clone so be prepared to wait while it pulls down files if you do this.

Generally a simple `git submodule update --init` is all that's needed in the SDK folder.

closed 2024-04-20 by earlephilhower

traxanos, 2024-04-20:
I want to test your changes and add this to pio.ini

platform_packages = framework-arduinopico @ https://github.com/earlephilhower/arduino-pico.git#d554df7

what were the better way?

earlephilhower, 2024-04-20:
Sorry, I don't use Platform.IO.  Check what @maxgerhardt wrote in the [docs here for P.IO](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst) for more info, it's pretty extensive.

traxanos, 2024-04-20:
ok i will enable long paths and use the old way :D that has works in past. thank you.