earlephilhower/arduino-pico #2129: Fix enum redeclaration issue when using WiFi and Ethernet

rpineau, 2024-04-27:
If we don't prevent the re-declaration of the enum this triggers
    a compiler  error as LwipIntfDev.h get included by both WiFi.h
    and the Ethernet modules (lwIP_w5500/src/W5500lwIP.h for example).
Without this, it's not possible to use the pico wifi and a W5500 

`diff --git a/libraries/lwIP_Ethernet/src/LwipIntfDev.h b/libraries/lwIP_Ethernet/src/LwipIntfDev.h
index 4bf7aab..3e548f7 100644
--- a/libraries/lwIP_Ethernet/src/LwipIntfDev.h
+++ b/libraries/lwIP_Ethernet/src/LwipIntfDev.h
@@ -50,11 +50,14 @@
 #define DEFAULT_MTU 1500
 #endif

+#ifndef __EthernetLinkStatus__
+#define __EthernetLinkStatus__
 enum EthernetLinkStatus {
     Unknown,
     LinkON,
     LinkOFF
 };
+#endif

 extern "C" void cyw43_hal_generate_laa_mac(__unused int idx, uint8_t buf[6]);`

earlephilhower, 2024-04-27:
Looks like you found a bug and a solution.  Want to make a PR?

rpineau, 2024-04-27:
Yes.
I guess I need to fork this first, make my change and make a PR from it ? (as I don't have commit rights).

JAndrassy, 2024-04-27:
sorry, but what error do you get? there is `#pragma once` to guard this

earlephilhower, 2024-04-27:
I just `grep`ped for that identifier and it's only defined in 1 spot in the entire tree.  In addition to @JAndrassy 's comments that the suggested diff doesn't do what you think it does, it also would break the `enum` typename.

````
earle@amd:~/Arduino/hardware/pico/rp2040$ grep -r EthernetLinkStatus
libraries/lwIP_Ethernet/src/LwipIntfDev.h:enum EthernetLinkStatus {
libraries/lwIP_Ethernet/src/LwipIntfDev.h:    EthernetLinkStatus linkStatus();
libraries/lwIP_Ethernet/src/LwipIntfDev.h:EthernetLinkStatus LwipIntfDev<RawDev>::linkStatus() {
````

Closing for now, but if you have an MCVE showing how it fails post it here and we'll look.

closed 2024-04-27 by earlephilhower

rpineau, 2024-04-27:
There is a #pragma once in it but if I mix it with the standard Ethernet library this doesn't prevent the redefinition.
Why do I use the standard Ethernet library with my W5500, because the code was already written before (by me) to use this for my project that I'm now porting to the pico W.

The diff above allows me to use the standard ethernet library, define `__EthernetLinkStatus__` (Github mangled the codfe and removed the double underscore around EthernetLinkStatus) and then include WiFi.h without having  EthernetLinkStatus redefine a 2nd time

rpineau, 2024-04-27:
I spent the last few hours moving all my code to Wiznet5500lwIPc for the W5500 I use (the WiFi code is disable for now).
Wiznet5500lwIPc Ethernet works fine for standard TCP connectivity, but UDP `begin` return 1
```
discoveryServer = new WiFiUDP();
nErr = discoveryServer->begin(m_UDPPort);
```
When using the standard Ethernet library with EternetUDP, the discovery works.
When changing to WiFiUDP , it gets the broadcast packet and replies, but the discovery service doesn't recognise the response.
tcpdump doesn't really show any difference in the payload so it's probably related to the failure on `begin`
Looking at the code for begin in `WiFiUdp.cpp` it uses a IPAddress() call to get the ip to listen on, but there is no way to set this IP (or I haven't found it) and I don't want to listen on 0.0.0.0.
The constructor for WiFiServer does take an IP and port as parameters, the constructor for WiFiUDP takes no parameters.
I'm for now only testing with the Ethernet side of my project as I can't move to Wiznet5500lwIPc + WiFi if I'm not able to get parity with what I have working with the standard Ethernet library.
May be the real problem I'm facing is how do I tie a server to an interface (on Linux or macOS I would bind to the respective interfaces IP to run services on different IP or 0.0.0.0 for all interfaces)

rpineau, 2024-04-28:
So after a lot more time spend on this, turn out the code works, but the lwIP_w5500 library is more than 20 times slower than the standard Arduino Ethernet library with a W5500.
The code implements a simple REST server and the exact same code returns a reponse in ~ 40ms with the standard Ethernet library versus slightly over a second with the lwIP_w5500 library. During the discovery after replying to the UDP broadcast, the service then query 3 of the API endpoint successively. As the lwIP_w5500 is slow to respond it fails the discovery. So the question is why is lwIP_w5500 so much slower ?

JAndrassy, 2024-04-28:
did you wire the INT pin?

rpineau, 2024-04-28:
No I didn't, but I can modify the PCB for this. 
Which pin should be used as the interrupt pin ?
The only pin I have left available on the PICO is GP27.
And is this configurable in the lwIP_w5500 library (is there a code example that shows how to set the pin).
If I need to use a specific pin I can probably change the board design (this is all open source in case you wonder, it's for an astronomy dome controller).

JAndrassy, 2024-04-28:
`ENCW5500lwIP Ethernet(CS_PIN, SPI, INT_PIN);`

rpineau, 2024-04-28:
Thanks (I did find this after digging in the code).
I'll wire my spare pin to the W5500 int_n and see if this helps with the speed (time to add that bodge wire ;) , it's mnot a proper prototype if there is not at least one bodge wire on it :) ).

rpineau, 2024-04-28:
So after wiring the INT_N pin of the W5500 chip (nothing like a bit of SMD soldering on a Sunday :) ) and declaring the ethernet object like this `Wiznet5500lwIP domeEthernet(ETHERNET_CS, SPI, 27);` ,  the query response time is now around  50ms and the discovery system now sees my board properly.
Thanks for all the help !

rpineau, 2024-04-28:
I'm ok modifying the PCB as I only made 2 prototypes for this round :

[https://sbforum.s3.amazonaws.com/DomeEthernetRP2040.jpg](url)

So for these 2, I just added the missing signal with a small wire and I have modified the PCB files to add it for the next prototype round.
I'd rather fix it the right way as using interrupt is so much more efficient than polling.
As this is for an open source / open hardware project , it's fine. It's not like I have to go fix a large number of existing board.

Thanks again for all the help.