earlephilhower/arduino-pico #2163: SdFat speed regression (because of pico SPI driver change)

FerGT50, 2024-05-17:
Hello,

with 3.8.1 I noticed a severe slowdown running SD Bench (included with SdFat 2.2.2).
With 3.3.0 (SdFat 2.1.1) I got 2.4 MB/s, now 930 KB/s.

Asked Bill Greiman (SdFat author) about that, he discovered a change in SPI Array Transfer:

https://github.com/greiman/SdFat/issues/481

Would it be possible to get array-transfer speed back?
Thanks!

Fernando

earlephilhower, 2024-05-17:
No changes to that section have been done in 3 years.  The speed regression is not related to an SPI change.  See the blame logs below:

https://github.com/earlephilhower/arduino-pico/blame/182af714924673059cccce88e296c762284d4a26/libraries/SPI/src/SPI.cpp#L136-L143

earlephilhower, 2024-05-17:
Reopening.  Not an SPI issue, but #801 and PR #819 show a patch I had to do to SdFat to get good speed.  We can check if that patch got undone or broken w/the 2.2.2 SdFat upstream...

FerGT50, 2024-05-17:
Thank you. You are always supportive. Arduino-pico is a game changer.

earlephilhower, 2024-05-17:
@FerGT50 please try the 1-line change in #2164 (you can just edit your SdFatConfig.h for simplicity) and let us know if that clears things up.  Looks like the array-transfer option setting got lost when I did the 2.2.2 merge.

FerGT50, 2024-05-17:
> @FerGT50 please try the 1-line change in #2164 (you can just edit your SdFatConfig.h for simplicity) and let us know if that clears things up. Looks like the array-transfer option setting got lost when I did the 2.2.2 merge.

Yes, Bill suggested the same quick fix in the meantime.
I can confirm it works: this is with cpu clock at 133 MHz and SPI clock at 62.5

```
FILE_SIZE_MB = 256
BUF_SIZE = 65536 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2324.76,65046,28088,28191
2324.69,64800,28090,28192

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2367.87,27723,27645,27678
2367.89,27718,27643,27678

Done
```

Many thanks!

PR #2164 (2024-05-17): Update SDFat to use array transfers
Fixes #2163
commit 83a6aaca (2024-05-17): Update SDFat to use array transfers (#2164)

Fixes #2163
closed 2024-05-17 by FerGT50