earlephilhower/arduino-pico #2198: New sketch not running with Arduino IDE OTA (Rpi Pico W)

jrrdev, 2024-06-05:
## Summary

When uploading a sketch through OTA with Arduino IDE, the process seems to end successfully. But when the Rpi Pico W reboots, the old sketch is still running, not the new one.
I can't see any errors during upload and in the debug logs (see details below)

## Steps to reproduce

* Board: Raspberry Pi Pico W
* Arduino IDE 1.8.19
* arduino-pico 3.9.1

* Arduino IDE board config:
[image]

* Sketch : BasicOTA example with this modifications:
  * STASSID and STAPSK defined to match my Wifi config
  * those lines are added in the end of `setup()` and commented / uncommented to see if the new sketch is running:
```c
pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH); 
```

1) Upload the sketch through Serial Port
2) After reboot, the Rpi is connected to Wifi, the LED is on and the Rpi is visible in "Network Ports" in the IDE
3) Comment the lines related to lighting up the LED
4) Upload sketch through Arduino IDE OTA
5) OTA process complete successfully. But after Rpi reboot, the LED is still on indicating that the old sketch is still running
6) Upload the sketch through Serial Port
7) After Rpi reboot, the LED is now off indicating that the new sketch is now running

## Logs

Arduino IDE:

```
[REDACTED PATH]/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 -I [REDACTED PATH]/packages/rp2040/hardware/rp2040/3.9.1/tools/espota.py -i [REDACTED IP] -p 2040 --auth= -f /tmp/arduino_build_231005/sketch_jun05a.ino.bin 
Uploading...................................................................................................................................................................................................................................................................................................
Complete
```

Serial Debug (only relevant parts, see [ota_serial_debug_rpi_pico.log](https://github.com/user-attachments/files/15594567/ota_serial_debug_rpi_pico.log) for full log):

```
OTA server at: pico-E66178758B699838.local:2040
Ready

IP address: [REDACTED IP]
(...)
[begin] _startAddress:     0x00000000 (0)
[begin] _size:             0x000676E8 (423656)
Start updating sketch
(...)
Progress: 100%
MD5 Success: 9ffe28c53125404842b4898770cba54b
lfs_file_close: fd=0x200126f4
lfs_file_close: fd=0x200126f4
:ack 3

lfs_file_close: fd=0x200126f4
Staged: address:0x00000000, size:0x000676E8
:urch 73, 45

:urd 2, 45, 0

:urd 1, 45, 2

:urd 1, 45, 3

:urd 2, 45, 4

:urd 2, 45, 6

:urd 2, 45, 8

:urd 2, 45, 10

:wr 2 0

:wrc 2 2 0

:ack 2

:rcl pb=0 sz=-1

:abort

Update Success

End

Rebooting...
OTA server at: pico-E66178758B699838.local:2040
Ready

IP address: [REDACTED IP]
```

earlephilhower, 2024-06-05:
Thanks for the detailed description.

I think this is due to the LittleFS update.  The on-flash format will be the changed 2.6.0+ version while the ota was built using the original flash format and won't be able to mount the filesystem.

jrrdev, 2024-06-05:
> @jrrdev can you give the rebuilt OTA in #2199 a try and report back? I'd like to do a hotfix release ASAP.

OK I will give it a try.
I can confirm that OTA is working in 3.9.0 so you must be right about the littleFS version problem

earlephilhower, 2024-06-05:
And to be clear, you need to flash using USB, not OTA, the first time on your Pico since it's already running the later LittleFS and in an inconsistent state.

jrrdev, 2024-06-05:
OTA is properly working after rebuilding the sketch with #2199 and uploading it first through USB.
Do you want me to perform some more tests ? For instance try a direct upgrade from 3.9.0 through OTA only ?

earlephilhower, 2024-06-05:
IF it's easy, that would be great.  But just thinking through an upgrade from 3.9.0 to the latest should be fine.

The 3.9.0 app will write to a LFS-old filesystem and reboot, the OTA will be overwritten as well as the app in the next one with both at LFS-2.6 format.  When the 3.9.2 app starts up it will migrate the FS to 2.6.0 format on flash, and the OTA already will have 2.6.0 support (and LFS-old since 2.9.3 can always read earlier LFS versions).

closed 2024-06-05 by earlephilhower

jrrdev, 2024-06-05:
You are right, I can confirm through test that upgrade from 3.9.0 through OTA works.
So the only users who will need to perform the upgrade through USB will be the ones who have upgraded to 3.9.1 first

PR #2199 (2024-06-05): Rebuild OTA bootloader for newer LittleFS version
The LittleFS folks changed the on-flash format in 2.6.0, making it unmountable with earlier precompiled code.

Rebuild the OTA bootloader using the 2.9.3 LittleFS release, matching the core version.

Fixes #2198
commit f08ba6bd (2024-06-05): Rebuild OTA bootloader for newer LittleFS version (#2199)

The LittleFS folks changed the on-flash format in 2.6.0, making
it unmountable with earlier precompiled code.

Rebuild the OTA bootloader using the 2.9.3 LittleFS release,
matching the core version.

Fixes #2198