earlephilhower/arduino-pico #2207: PICO W STUCK DUE TO FLASH CORRUPTION.

JITU-SYS, 2024-06-07:
In our code, we are using the arduino littlefs library to log the error codes , and keep the few device configurations

We are passing the device_info structure to the functions to log the important information into the NV-ROM

We have created the MACRO called "SET_ERROR(error_type)" to log the hex error code into the littlefs file system during the run time if any error occurs
After some observations, we know that the code might be stuck while storing the error code in the FS

Every time, it opens the deviceInfo.txt file, log the error, and then closes it.

I have attached the MACROS and WriteToFile functions snippets here

Previously, we used the same logic with the many controllers from different manufacturers, but this is the very first time with the RP2040
Based on that, I have a couple of questions
1. how reliable it is to use the little to keep the error codes, and is there any chance of flash corruption because of read and write cycles?
2. Does Little Library have any blocking functions or internal calls at the lower-level drivers?
3. Does opening the file and closing it cause any issues or any chance of flash corruption?
4.
Let me know if you have any suggestions for the same
Appreciate all suggestions and comments

PS: we have allocated 1MB to sketch and 1MB to FS

earlephilhower, 2024-06-07:
Without a small MCVE there's nothing anyone can do to help you with such a vague problem, unfortunately.  I really don't understand your questions, but you can probably answer them yourself by going through the code provided here in the core, it's all there.

As far as LittleFS and flash corruption, unless you're hitting the limits of your flash chip (and its physically failing), LFS was designed to be very reliable and is in use by hundreds of projects here and probably thousands of embedded applications around the world.

closed 2024-06-07 by earlephilhower

JITU-SYS, 2024-06-08:
Hey @earlephilhower thanks for replying there, than please guide me with this, as my raspberry pi pico w is going under hardfault/ARM M in lockup state. What should be the behaviour of rp2040 after this ? 

Are there any hardfault handler implemented in pico-sdk v1.5.1 ?
As I have looked inside the sdk files and have read forums , there is no mention of hardfault handler.

Your comments and suggestions are really helpful in this case .

Below are the images showing you the insights of RP2040 going into hardfault / ARM lockup state .

[image]