earlephilhower/arduino-pico #2231: Bluetooth LE Notifications not working

ullibak, 2024-06-17:
First of all, thanks to Earle for creating and maintaining this beautiful library!

I have a device that sends data through BLE notifications. I used the example BTStack -> LECentral and modified it for the service and characteristic UUIDs of my device. I was able to connect and read services and characteristics. I was also able to subscribe to notifications and the status value of the gattSubscribedCallback() function suggests that the notification was successfully registered. But the gattCharacteristicNotification() callback routine was never triggered when my device sent data. 

I then compared to the BLE client and server temperature measurement example in the Pico SDK 
(\pico-examples\pico_w\bt\standalone\server.c and client.c) and found that maybe something is missing in the subscribeForNotifications() function in BTStackLib.cpp

After I modified this function, the notification callback routine was triggered and everything seemed to work as expected. 

```c++
int BTstackManager::subscribeForNotifications(BLEDevice * device, BLECharacteristic * characteristic) {
    gattAction = gattActionSubscribe;
    // ADDED
    static gatt_client_notification_t notification_listener;
    gatt_client_listen_for_characteristic_value_updates(&notification_listener, gatt_client_callback,  device->getHandle(), (gatt_client_characteristic_t*) characteristic->getCharacteristic());
    // END ADDED
    return gatt_client_write_client_characteristic_configuration(gatt_client_callback, device->getHandle(), (gatt_client_characteristic_t*) characteristic->getCharacteristic(),
            GATT_CLIENT_CHARACTERISTICS_CONFIGURATION_NOTIFICATION);
}
``` 
This is probably not the right way to fix this issue, but could maybe be a hint to a proper solution. The same solution should then also be applied to the subscribeForIndications() function and maybe also the corresponding unsubscribe functions.

Many thanks!

earlephilhower, 2024-06-17:
Thanks for the debug and proposed solution!  Why not throw together a PR with your suggestion, since you have a use case and came up with the fix?  It may not be perfect, but if what is there now doesn't work (and the BTStack guys haven't kept up with their Arduino library so it's not going to be fixed upstream) we aren't likely to be worse off.

earlephilhower, 2024-06-18:
Looking at the code, I think you're spot on with the fix, but the `static` local variable in a member function will get overwritten by every call to `subscribeForXXX`.  Some sort of `BLECharacteristic` <-> `gatt_client_notification_t` map needs to be tracked,  or `malloc`'d on every subscription to keep it from trouncing priors...

ullibak, 2024-06-18:
In the meantime, I made some modifications that might better fit to the structure of BTstacklib.

I reverted the subscribeForNotifications function to its original version and added an additional function called listenForCharacteristicValueUpdates. 

To do this, I had to add 

```c++
int BLEDevice::listenForCharacteristicValueUpdates(BLECharacteristic * characteristic){
    return BTstack.listenForCharacteristicValueUpdates(this, characteristic);
``` 
and

```c++
int BTstackManager::listenForCharacteristicValueUpdates(BLEDevice * device, BLECharacteristic * characteristic) {
    gatt_client_listen_for_characteristic_value_updates(&notification_listener, gatt_client_callback,  device->getHandle(), 
            (gatt_client_characteristic_t*) characteristic->getCharacteristic());
    return 1;
``` 
to BTstacklib.cpp as well as the corresponding function definitions to BTstacklib.h

To circumvent the problem with the local variable, I added the definition

```c++
static gatt_client_notification_t notification_listener;
``` 
at the beginning of BTstacklib.cpp

In the ino-file (derived from the LECentral example), the new function is called before the original subscription function:

```c++
device->listenForCharacteristicValueUpdates(&characteristics[charMeasurement].characteristic);
device->subscribeForNotifications(&characteristics[charMeasurement].characteristic);
``` 

It works, but I have to state that I am by no means a proficient programmer nor am I an expert on BLE. This code was derived from a simple "copycat"-approach. Particularly, I do not understand the bookkeeping with the variable "notification_listener". In the example of the pico SDK, a function named 
```c++
gatt_client_stop_listening_for_characteristic_value_updates(&notification_listener);
``` 
is called at disconnect. My code works without this, but I suppose that there is a reason for calling this function. 
My programming knowledge does not allow me to implement this function in a senseful way, so help would be appreciated...

ullibak, 2024-06-21:
Found that the wrapper has been derived from the Arduino port that BlueKitchen published (a long time ago...). 

Therefore, posted the issue there (https://github.com/bluekitchen/btstack/issues/615) hoping that they know exactliy how to implement the subscription for notifications correctly.

gonzob, 2024-06-22:
Thanks also to Earle. I've been tearing my hair out over this exact problem for the last few days. I'm trying to build a rpi pico central for MIDI messages from an instrument. With the LECentral code it would connect, then nothing...
I confirm that Ullibak's original post works for me, and I await any developments.
Thanks.
Gonzo

earlephilhower, 2024-06-22:
Can either of you post a simple example that I can try locally without and with the fixes?  I'm fine doing the PR here, since the Bluestack Arduino library is basically abandoned (but mostly working!).

gonzob, 2024-06-22:
As i mentioned above my application is BLE MIDI. Here's my modified LECentral code:

`#include <BTstackLib.h>
#include <SPI.h>

/*
   EXAMPLE_START(LECentral): LE Central

   @text Compared with the other examples, the LE Central is
   a bit more complex. This is because it performs multiple
   steps in sequence as it is common with GATT Client APIs.

   It shows how to first scan for other
   devices and then connect to one. When connected, a series of
   GATT Client operations are performed: first the list of
   GATT Services is queried. If a particular service is found,
   the list of its GATT Characteristics is retrieved and a set
   of known Characteristics are cached for later access.
*/

/*
   @section Characteristic Summary
   @text As multiple Characteristics need to be found, a custom
   struct is used to collect all information about it. This allows
   to define the list of necessary characteristics in the
   characteristics[] array
*/
/* LISTING_START(LECentralSummary): Characteristic Summary */

// BLE Shield Service V2 incl. used Characteristics
//UUID bleShieldServiceV2UUID("B8E06067-62AD-41BA-9231-206AE80AB550");
UUID bleShieldServiceV2UUID("03b80e5a-ede8-4b33-a751-6ce34ec4c700");

typedef struct characteristic_summary {
  UUID         uuid;
  const char * name;
  bool         found;
  BLECharacteristic characteristic;
} characteristic_summary_t;

typedef enum characteristicIDs {
  charRX = 0,
  charTX,
  charBaud,
  charBdAddr,
  numCharacteristics  /* last one */
} characteristicIDs_t;

//characteristic_summary characteristics[] = {
//  { UUID("f897177b-aee8-4767-8ecc-cc694fd5fcee"), "RX", false, BLECharacteristic() },
//  { UUID("bf45e40a-de2a-4bc8-bba0-e5d6065f1b4b"), "TX", false, BLECharacteristic() },
//  { UUID("2fbc0f31-726a-4014-b9fe-c8be0652e982"), "Baudrate", false, BLECharacteristic() },
//  { UUID("65c228da-bad1-4f41-b55f-3d177f4e2196"), "BD ADDR", false, BLECharacteristic() }
//};

characteristic_summary characteristics[] = {
  { UUID("7772e5db-3868-4112-a1a9-f2669d106bf3"), "MIDI", false, BLECharacteristic() }
};

/* LISTING_END(LECentralSummary): Characteristic Summary */

// Application state
BLEDevice  myBLEDevice;
BLEService myBLEService;
bool serviceFound;
bool sendCounter = false;

int counter = 0;
char counterString[20];

[... 265 lines cut ...]
  // then save the bytes, ignoring timing bytes
  int j = 0; // count the number of bytes
  for (int i = 2; i < length;){   
      Serial.print((byte)value[i]); // Status byte
      i += 1;
      j += 1;
      // detect the timing byte
      while ((byte)value[i] < 128 and i < length and j < 15){ 
        Serial.print((byte)value[i]);
        i += 1;
        j += 1;
      }
      i +=1; //skip the timing byte
    }
    Serial.println("");

}
/* LISTING_END(LECentralNotificationCallback): Notification Callback */`

To test it you'll need a MIDI perpheral such as a keyboard, or wind controller. The notification callback is a work in progress so it's not very flash...

What I'm intending to do wit this project is to receive BLE MIDI from an instrument (wind controller) and pump it into a synth which has a USB port. This means that the microcontroller needs to be able to present a USB-MIDI port to the synth. 
I originally tried to do this in MicroPython (because I know it can do the USB-MIDI) but I couldn't get any notifications from the instrument to appear. In frustration I swapped to the Arduino IDE, but had the same problem. After I got it working last night (Thankyou!), I then switched to getting the USB-MIDI port working, and <AAAAGH> it appears that there's no code for that.....

Gonzo

ullibak, 2024-06-23:
Hi
I can post my example that I used for debugging. I started with the example of a temperature sensor found in the example directory of the pico SDK: C:\pico\pico-examples\pico_w\bt\standalone and compiled it using VSCode.

The example code "server.c" sends a temperature measurement derived from an internal diode every couple of seconds through BLE notification. I modified this code (and the makefile) to send a temperature reading every 5 sec. and also output the temperature measurement and the device's address via USB. 
To get the device running, simply copy the UF2 file in this zip archive to the pico and hook it up to some terminal at 115200 bd, e.g. the Serial Monitor of the Arduino IDE:
[picow_ble_temp_sensor.zip](https://github.com/user-attachments/files/15943959/picow_ble_temp_sensor.zip)

The output should be something like

```
Write temp 19.55 degc
sent from address D8:3A:DD:20:70:07
```
repeated every 5 seconds.

Then, take a second Pico device and copy the code below (derived from the LECentral example) to the arduino IDE. Change the address in the variable addrString to the address that the first Pico showed. Compile, upload and look at the output of the Serial Monitor.
The program looks at all the advertisments of BLE devices and if the address of the Pico temperature sensor is found, it connects and subsctribes to notifications.

```c++
#include <BTstackLib.h>
#include <SPI.h>
#include <time.h>

typedef struct characteristic_summary {
  UUID         uuid;
  const char * name;
  bool         found;
  BLECharacteristic characteristic;
} characteristic_summary_t;

typedef enum characteristicIDs {
  charMeasurement = 0,
  numCharacteristics  
} characteristicIDs_t;

// Enter address of Pico temperature sensor here:
char* addrString = "D8:3A:DD:20:70:07"; 

UUID myServiceUuid("0000181A-0000-1000-8000-00805F9B34FB"); // Micropython temperature example from Pico SDK

characteristic_summary characteristics[] = {
  { UUID("00002A6E-0000-1000-8000-00805F9B34FB"), "Measurement", false, BLECharacteristic()} // Micropython temperature example from Pico SDK
};        

time_t now, old_time;

BLEDevice  myBLEDevice;
BLEService myBLEService;
const int maxServices = 50; //max. 50 services should be enough
int numServicesFound = 0; //cout number of services found
BLEService foundServices[maxServices]; //max. 50 services should be enough

bool serviceFound;
bool characteristicFound = false;
bool sendCounter = false;
bool LED_status = false;

static gatt_client_notification_t notification_listener;

void setup(void) {
  Serial.begin(115200);
  while(!Serial){}; // wait for serial connection. Remove for real application...
  pinMode(LED_BUILTIN, OUTPUT);
  BTstack.setBLEAdvertisementCallback(advertisementCallback);
  BTstack.setBLEDeviceConnectedCallback(deviceConnectedCallback);
  BTstack.setBLEDeviceDisconnectedCallback(deviceDisconnectedCallback);
  BTstack.setGATTServiceDiscoveredCallback(gattServiceDiscovered);
  BTstack.setGATTCharacteristicDiscoveredCallback(gattCharacteristicDiscovered);
  BTstack.setGATTCharacteristicNotificationCallback(gattCharacteristicNotification);
  BTstack.setGATTCharacteristicIndicationCallback(gattCharacteristicIndication);
[... 308 lines cut ...]
Device connected!
Service Discovered. UUID String:1800, UUID128 String:00001800-0000-1000-8000-00805F9B34FB
Service Discovered. UUID String:1801, UUID128 String:00001801-0000-1000-8000-00805F9B34FB
Service Discovered. UUID String:181a, UUID128 String:0000181A-0000-1000-8000-00805F9B34FB
Service located
Service discovery finished
Requested Service found. Search for Characteristics...
Characteristic Discovered: UUID String:2a6e, UUID128 String: 00002A6E-0000-1000-8000-00805F9B34FB, handle: 0x9, properties: 50
Characteristic found: Measurement
Characteristic discovery finished, status 1
Subscribe to Notifications
In gattSubscribedCallback(). Status: 0, Handle: 0x64
```
This means that the device with the given address has been found, connection has been established and the subscription to notifications was successful (status = 0). 

Without any modifications to BTstack.cpp, the output stops there and the notification callback routine, that should output the temperature value, is never called.

With the modifications described above (i.e. adding the _gatt_client_listen_for_characteristic_value_updates()_ function in some or the other way, the notification works as expected and the temperature is printed out every 5 seconds:

```
Notification received. Length: 2, handle: 0x9 Temperature 17.68 degc
Notification received. Length: 2, handle: 0x9 Temperature 17.21 degc
Notification received. Length: 2, handle: 0x9 Temperature 17.21 degc
Notification received. Length: 2, handle: 0x9 Temperature 18.61 degc
```

earlephilhower, 2024-06-23:
Awesome, thanks for both the samples.  I think with one characteristic, like you're running @ullibak , the `static` is fine, but if you added a 2nd one the BLE infrastructure will get confused.  Let me poke around and see the best place to stuff that extra housekeeping needed.

Also, FWIW, the BTStack guys seem to say that the library code as-is is correct, w/o the add'l `gatt_client_listen_for_characteristic_value_updates` call.  See their [posting here](https://groups.google.com/g/btstack-dev/c/WpulAkOiI9Y/m/YLCg1sTyAwAJ).  Not sure of what to make of it, but maybe things in the BTStack core have changed since their original posting.  And if it works with the add'l call, I sure don't mind adding it. :)

ullibak, 2024-06-24:
Thank you Earle! 

In the link that you posted, Matthis writes (quote):

"Anyway, it's enough to call gatt_client_write_client_characteristic_configuration to enable notifications, no need to gatt_client_write_characteristic_descriptor manually.

**To get notifications, you also have to register a listener for that characteristic with gatt_client_listen_for_characteristic_value_updates**."

As I understand this, the gatt_client_listen_for_characteristic_value_updates() function is mandatory to get notifications.

Bluekitchen also uses this function e.g. in the example
https://github.com/bluekitchen/btstack/blob/2b49e57bd1fae85ac32ac1f41cdb7c794de335f6/example/gatt_heart_rate_client.c#L178

just before subscribing to notifications. I suppose that it simply was forgotten when BlueKitchen created their Arduino port.

earlephilhower, 2024-06-24:
#2241 adds that new structure to every characteristic and uses it in the listen/unlisten methods.  I could not find any calls that look similar for BLE indicators, and there are no instances of their use in the BTStack examples, so I have not added anything to them.  Tested w/the BLE temp sensor MCVE posted here.

ullibak, 2024-06-24:
@Earle: I see that you are already working on the solution. In the meantime, Matthias gave an input on how to implement a "general" listener:
https://github.com/bluekitchen/btstack/issues/615

PR #2241 (2024-06-24): Receive GATT characteristic updates in BTStackLib
Fixes #2231
commit 1357e4a3 (2024-06-24): Receive GATT characteristic updates in BTStackLib (#2241)

Fixes #2231
closed 2024-06-24 by earlephilhower