earlephilhower/arduino-pico #2247: Support for BLE extended advertising?

jhmaloney, 2024-06-28:
Does the current BTstack library support Bluetooth 5.0 extended advertising (i.e. over 31-bytes of advertising data)?

From looking at the code in hci.c, it appears that the limit is 31 bytes.

If the BTstack library is based the Infineon port (https://github.com/Infineon/btstack/releases), it appears that support for extended advertising got added in the "BTSTACK v3.7.1" release of May 8, 2023 (https://github.com/Infineon/btstack/releases/tag/release-v3.7.1).

Is there any way to find out the version of the BTstack library in arduino-pico?

Do you have a timeline (or guess) for when you will update the BTstack library? (I realize you are probably forking the BTstack library from elsewhere (Arduino?) and don't have control over when they update.)

I also realize that arduino-pico BLE support is still in alpha.

 I'm not desperate for this feature. I just want to be sure that extended advertising is really not supported yet so I can stop looking for it. :-)

Thanks!

earlephilhower, 2024-06-28:
You're linking to a completely different BTStack than what the Pico W uses.  It looks proprietary to the Infineon chips and seems to just be a binary blob on GH without source to try and port elsewhere.  The API is definitely nothing like what we have here.

This is the [BTstack we use](https://github.com/bluekitchen/btstack/tree/72ef1732c954d938091467961e41f4aa9b976b34), directly from the RPI Pico SDK.  There is no extended advertising AFAICT with the version in the SDK.  You can check their `master` branch and see if it's something they have added, but that would not show up here until the RPI folks merge it in and do a release of the SDK.

closed 2024-06-28 by earlephilhower

jhmaloney, 2024-06-28:
Thanks for the quick reply!

Actually, I am linking to your version of BTstack (pico-sdk/lib/btstack), I just didn't know where it came from. It has been working fine so far except for the lack of extended advertising, which I can live without for now.

Thank you for verifying that extended advertising is not yet in the version we are using.

FYI, MicroBlocks is using the NimBLE-Arduino BLE stack (https://github.com/h2zero/NimBLE-Arduino) on the Nordic and Espressif chips we support. It's quite solid and has support for the most modern Bluetooth 5 features such as extended advertising. Unfortunately, it doesn't support the Raspberry Pi Pico W, so I'm really glad that you incorporated BTstack into this project.

Thank you  for all your hard work on Arduino-Pico.

earlephilhower, 2024-06-28:
I just re-checked and there does seem to be support for extended advertising in the SDK version of BTStack, but it was not enabled in the SDK or the config file here.  However, it is possible to edit `tools/libpico/btstack_config.h` and add `#define ENABLE_LE_EXTENDED_ADVERTISING` and rebuild the library.

Actually handling/serving those events would require you to write the low-level BTstack packet handlers.  The C++ wrapper from BTStack (libraries/BTStackLib) does not have any support for them.

jhmaloney, 2024-06-28:
Thanks! That's useful info; it means the Bluetooth hardware can handle extended advertisements.

I'm not desperate for that feature nor eager to dig into the low-level BTstack packet handler code. So I will hope that the RPI Pico SDK folks will do that at some point. :-)