earlephilhower/arduino-pico #2258: OTA Sketch Upload

djaus2, 2024-07-09:
Taking this as wishlist, would it be possible to be able to do Over- The-Air updates of a Sketch? 
Like [picowota](https://github.com/usedbytes/picowota)
Assuming an initial flash required by USB then updates OTA just of the Sketch.

djaus2, 2024-07-09:
I see there are some closed issues on OTA.
Looking into these to see if what I want is doable already with RPi Pico W

Frogieder, 2024-07-09:
You may be asking for something that's already been implemented here, be sure to check out [https://arduino-pico.readthedocs.io/en/latest/ota.html](https://arduino-pico.readthedocs.io/en/latest/ota.html).

earlephilhower, 2024-07-09:
OTA is supported on any Pico.  See the docs or examples(PicoOTA, ArduinoOTA, and the HTTPUpdate libraries in particular).

closed 2024-07-09 by earlephilhower

djaus2, 2024-07-09:
Thanks

David Jones (MVP)

djaus2, 2024-07-10:
OK got that to work, the BasicOTA example. :)