earlephilhower/arduino-pico #2261: EEPROM example

SeyranGrig, 2024-07-12:
~\Arduino ide\EEpromTest\EEPROM.cpp: In member function 'bool EEPROMClass::commit()':
~\Arduino ide\EEpromTest\EEPROM.cpp:122:5: error: 'rp2040' was not declared in this scope
     rp2040.idleOtherCore();
     ^~~~~~
D:\YandexDisk\YandexDisk\Yavchunovsky\PiPicoEthernet\Arduino ide\EEpromTest\EEPROM.cpp:122:5: note: suggested alternative: 'p24'
     rp2040.idleOtherCore();
     ^~~~~~
     p24

exit status 1

Compilation error: 'rp2040' was not declared in this scope

SeyranGrig, 2024-07-12:
EEPROM.cpp:141: undefined reference to `_EEPROM_start'

earlephilhower, 2024-07-12:
You are not using this core, so you can't take bits out of it like the EEPROM library and expect it to work w/the Arduino.cc core.  I suggest you look at their own options for this, maybe a filesystem instead.

closed 2024-07-12 by earlephilhower