earlephilhower/arduino-pico #2275: Feature Request: Increased Joystick sensitivity

Jediknite1101, 2024-07-18:
Let me be first to say I am a novice when it comes to microcontrollers and the Arduino IDE.  Is it possible to increase the joystick axis sensitivity from 8bit to 16bit? If so is this something that I could possibly modify if I were to branch? I'm building an ASC with high sensitivity encoders that will map to the joystick axis.

Realrobots created a joystick HID library for the official Arduino Pico board manager with those 16bit ranges. Which honestly I'd love to be able to use with your Pico board profile, but there are few core lib differences that prevent his lib from compiling.

earlephilhower, 2024-07-18:
It's kind of trivial to turn the gamepad into a 16-bit device, but the current TinyUSB doesn't have helper functions/macros to do so like it does for the 8-bit mode implemented now.  A trivial implementation w/Adafruit_TinyUSB can be found at https://gist.github.com/OrhanYigitDurmaz/a37b410e4875a625a515a4def4397046 .  Basically the HID descriptor needs to say each axis is 16, not 8 bits.

I'll need to think of a way to make this work w/the existing Joystick.  Doing a parallel `Joystick16` class would be simplest but have too much code duplication for my liking.  But, the 8- or 16-bit choice needs to be determined at compile time since the USB descriptor is created in `main()` before `setup()` is ever called...

closed 2024-07-18 by earlephilhower

Jediknite1101, 2024-07-24:
Thank you so much for the update. I appreciate your dedication to the upkeep of your board profiles.

earlephilhower, 2024-07-24:
No worries!  Your "Thank You" note actually made me realize I needed to update the BT/BLE version as well, thanks! (#2293)

commit 62e16c02 (2024-07-18): Add true 10- and 16-bit joystick modes

Fixes #2275

Adds `Joystick.use10bit` and `Joystick.use16bit` methods.  10-bit is
unsigned from 0...1023 while 16-bit is signed -32767..32767.
Defines a new HID descriptor to support the increased resolution.
PR #2276 (2024-07-18): Add true 10- and 16-bit joystick modes
Fixes #2275

Adds `Joystick.use10bit` and `Joystick.use16bit` methods.  10-bit is unsigned from 0...1023 while 16-bit is signed -32767..32767. Defines a new HID descriptor to support the increased resolution.
commit bd902a93 (2024-07-18): Add true 10- and 16-bit joystick modes (#2276)

Fixes #2275

Adds `Joystick.use10bit` and `Joystick.use16bit` methods.  10-bit is
unsigned from 0...1023 while 16-bit is signed -32767..32767.
Defines a new HID descriptor to support the increased resolution.