earlephilhower/arduino-pico #2303: error: '__lockBluetooth' was not declared in this scope - when trying to compile     BluetoothHIDMaster/examples  /KeyboardPiano.ino

cyberboy666, 2024-07-27:
i get the following error when trying to compile the example [KeyboardPiano.ino](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/BluetoothHIDMaster/examples/KeyboardPiano/KeyboardPiano.ino):

```
'__lockBluetooth' was not declared in this scope
```

<details><summary>full stack trace of error</summary>

```
In file included from /home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothHCI.cpp:27:
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h: In constructor 'BluetoothLock::BluetoothLock()':
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h:29:9: error: '__lockBluetooth' was not declared in this scope
   29 |         __lockBluetooth();
      |         ^~~~~~~~~~~~~~~
Compiling .pio/build/pico/FrameworkArduino/RP2040Support.cpp.o
Compiling .pio/build/pico/FrameworkArduino/RP2040USB.cpp.o
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h: In destructor 'BluetoothLock::~BluetoothLock()':
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h:32:9: error: '__unlockBluetooth' was not declared in this scope
   32 |         __unlockBluetooth();
      |         ^~~~~~~~~~~~~~~~~
Compiling .pio/build/pico/FrameworkArduino/SerialPIO.cpp.o
In file included from /home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHIDMaster/src/BluetoothHIDMaster.cpp:64:
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h: In constructor 'BluetoothLock::BluetoothLock()':
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h:29:9: error: '__lockBluetooth' was not declared in this scope
   29 |         __lockBluetooth();
      |         ^~~~~~~~~~~~~~~
Compiling .pio/build/pico/FrameworkArduino/SerialUART.cpp.o
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h: In destructor 'BluetoothLock::~BluetoothLock()':
/home/timcaldwell/.platformio/packages/framework-arduinopico/libraries/BluetoothHCI/src/BluetoothLock.h:32:9: error: '__unlockBluetooth' was not declared in this scope
   32 |         __unlockBluetooth();
      |         ^~~~~~~~~~~~~~~~~
Compiling .pio/build/pico/FrameworkArduino/SerialUSB.cpp.o
*** [.pio/build/pico/lib854/BluetoothHCI/BluetoothHCI.cpp.o] Error 1
Compiling .pio/build/pico/FrameworkArduino/StackThunk.cpp.o
*** [.pio/build/pico/libb6d/BluetoothHIDMaster/BluetoothHIDMaster.cpp.o] Error 1
```

</details>

im using platformio btw with `platformio.ini`:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 1m
platform_packages = 
	framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#master
monitor_speed = 115200
build_flags = 
 	-DPIO_FRAMEWORK_ARDUINO_ENABLE_BLUETOOTH
```

im wondering ig theres something small im missing ?

also as an aside, my goal here is to use a _pi pico w_ to convert a __bluetooth (classic) HID keyboard__ to a __USB HID keyboard__ - is this likely to be possible using this framework ? 

thanks again for your work on this project

earlephilhower, 2024-07-27:
Set the board in your ini to `rpipicow`.  The `pico` doesn't have BT.

closed 2024-07-27 by earlephilhower

cyberboy666, 2024-07-27:
ahh thank you @earlephilhower ! was something obvious i was missing doh !