earlephilhower/arduino-pico #2328: LittleFS unable to create a file when FS is uploaded

HamzaHajeir, 2024-08-11:
Hi there,

In my journey to fully port a framework, I've faced an issue related to the FS, wherein the FS could not create a file if the system has already at least two files, whether being uploaded or created dynamically.

MCVE: 
```cpp
#include <Arduino.h>
#include <LittleFS.h>
#include <string>

void writeFile(const char* fn, const char* content) {
	File file = LittleFS.open(fn, "w+");
	file.write(content);
	file.close();

}
std::string readFile(const char* fn) {
	File file = LittleFS.open(fn, "r");
	if (!file) {
		Serial.printf("Failed to open\n");
		return "";
	}
	std::string data (file.size(), '#');
	file.readBytes(data.data(),file.size());
	file.close();
	return data;
}

void checkFile(const char* fn) {
	auto data = readFile(fn);
	Serial.printf("File [%s]: ", fn);
	if (data.empty()) {
		Serial.printf("[EMPTY/INEXISTENT]\n");
		return;
	}
	Serial.printf("\"%s\"\n", data.c_str());
}
void setup() {
	Serial.begin(921600);
	delay(5000);
	if (!LittleFS.begin()) {
		Serial.printf("Failed to mount LittleFS\n");
	}
	if (!LittleFS.format()) {
		Serial.printf("Failed to format LittleFS\n");
	}
	writeFile("/somefile", "Test1");
	checkFile("/somefile");

	writeFile("/somefile2", "Test2");
	checkFile("/somefile2");

	writeFile("/somefile3", "Test3");
	checkFile("/somefile3");
}
void loop() {

}
```

Output:

```
lfs_file_close: fd=0x20011a34
lfs_file_close: fd=0x20011a34
File [/somefile]: "Test1"
lfs_file_close: fd=0x20011a34
Unable to set last write time on '/somefile2' to 5
lfs_file_close: fd=0x20011a34
File [/somefile2]: "Test2"
LittleFSDirImpl::openFile: rc=-36 fd=0x20011a34 path=`/somefile3` openMode=5 accessMode=3 err=-36
LittleFSDirImpl::openFile: rc=-2 fd=0x20011a34 path=`/somefile3` openMode=0 accessMode=1 err=-2
Failed to open
File [/somefile3]: [EMPTY/INEXISTENT]
```

platformio.ini: 

```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
framework = arduino
board_build.core = earlephilhower
monitor_speed = 921600

lib_ldf_mode = deep+

build_flags = -DEMBEDDED_PLATFORM
	'-DARDUINO_BOARD="RASPBERRY_PI_PICO_W"'
	-DDEBUG_RP2040_PORT=Serial

board_build.filesystem_size = 0.6m

upload_port = F:
```

earlephilhower, 2024-08-11:
It works fine from Arduino IDE.

````
File [/somefile]: "Test1"
File [/somefile2]: "Test2"
File [/somefile3]: "Test3"
````

My guess is you're specifying something wrong in the `platform.ini`.  Try using `0.5m` for the filesystem size instead of `0.6m`.  512KB is a standard size, 0.6M isn't and I bet it's not actually implemented.

HamzaHajeir, 2024-08-12:
> My guess is you're specifying something wrong in the `platform.ini`. Try using `0.5m` for the filesystem size instead of `0.6m`. 512KB is a standard size, 0.6M isn't and I bet it's not actually implemented.

Correct, it was that configuration, which I though it was easily editable.

Thank you, now it works as expected.

closed 2024-08-12 by HamzaHajeir