earlephilhower/arduino-pico #2330: Support for or alternative to Scheduler.h API?

pomplesiegel, 2024-08-11:
Hello! I am transitioning from using a RP2040 on PlatformIO using "pico" and the standard core to `arduino-pico` instead. 

Question: I was previously using `Scheduler.h`, which has built-in support on the standard PlatformIO architecture, but I can't seem to get it working with `arduino-pico`. Is there a way to get it working, or a drop-in alternative with similar behavior and/or API? 

Thank you!

earlephilhower, 2024-08-11:
Scheduler uses MBEDos threading.  It's not something that can run on bare metal like this core is.

What you can do instead is look at `FreeRTOS` which is supported (in SMP mode, so both cores used automatically) and has support for `task`s like this.

closed 2024-08-11 by earlephilhower

pomplesiegel, 2024-08-11:
Great, thank you!