earlephilhower/arduino-pico #2338: W6100-EVB-Pico Support?

DavidGoedicke, 2024-08-18:
Hello I saw that  library already support the  W5100-EVB-PICO and W5500-EVB-PICO and was wondering if there is any support planned for the W6100-EVB-PICO.

As far as I can tell it's "just" a few different address registers but it is just above my head to make these changes myself.

I am super to help support and debug in any way I can, and I 'd really appreciate your help.

[W6100-EVB-PICO](https://docs.wiznet.io/Product/iEthernet/W6100/w6100-evb-pico)
[W6100-chip](https://docs.wiznet.io/Product/iEthernet/W6100)

sstaub, 2024-08-18:
There is a branch for the 6100 on the WIZnet Github page. It is some years old.
https://github.com/Wiznet/Ethernet/tree/W6100

earlephilhower, 2024-08-18:
Note that the w6100 provides lots of new services, but we wouldn't use any of it *at al*l.

All the IP stack lives inside the Pico.  The Ethernet adapters are simply Ethernet packet send/receive units and any intelligence in them is not used.  A W5100/W5500 may be less expensive and give the same functionality here (we support IPV4/V6 natively on all Ethernet devices, for example).

DavidGoedicke, 2024-08-20:
Thanks for the response! I will go and find the W5100/W5500 boards alternatively. 

Thanks for the great tool! 

I think we can close this issue for now?!

closed 2024-08-20 by earlephilhower