earlephilhower/arduino-pico #2347: FatFSUSB plug() not invoked

soundstorm, 2024-08-22:
I've tried to use the Listfiles-USB sketch from the examples, works quite well. Except the `plug()`-callback is not invoked.
Tried to print the `driveConnected` in the loop, but always reports 0. I've tried to implement a `isConnected()` method, based on a `_isConnected` variable, that should change within the `plug()` and `unplug()` methods withing FatFSUSB, but sadly not only the callback is not called, neither the main `plug()` method.

earlephilhower, 2024-08-22:
Without an MCVE that reproduces the problem, there's really not much anyone can do.  Can you provide a small, complete example showing the problem?

soundstorm, 2024-08-22:
I've just used the Listfiles-USB sketch and added a `Serial.println(driveConnected);` every second to the main loop, which always yields 0. Using a regular Pi Pico.
What further information do you need?

soundstorm, 2024-09-24:
Was there any change in the code or just closed?

earlephilhower, 2024-09-24:
FF had to restart after an update this morning and looks like it lost the message, sorry.  The TinyUSB call is registered and does get called on my Ubuntu install, but depending on the state of the moon Windows may or may not actually send the USB packet that TUSB is looking for on mounting.  So this is out of the core's hands, but it's hooked into the proper frameworks.  There might be something you could push upstream to TInyUSB, but they're really just a thin wrapper as well...