earlephilhower/arduino-pico #2351: LittleFS: removing a directory removes all the directories along the path

jlbirccyn, 2024-08-24:
Hello,

I ran into a problem in LittleFS. Doing an rmdir deletes all directories along the path, not just the deepest one. 

Here's a sketch that shows the problem:

```C
#include <LittleFS.h>

void listAllFilesInDir(String dir_path) {
  Dir dir = LittleFS.openDir(dir_path);
  while (dir.next()) {
    if (dir.isFile()) {
      // print file names
      Serial.print("File: ");
      Serial.println(dir_path + dir.fileName());
    }
    if (dir.isDirectory()) {
      // print directory names
      Serial.print("Dir: ");
      Serial.println(dir_path + dir.fileName() + "/");
      // recursive file listing inside new directory
      listAllFilesInDir(dir_path + dir.fileName() + "/");
    }
  }
}

void setup() {
  Serial.begin(115200);
  while (!Serial)
    ;

  LittleFS.begin();

  Serial.println("-1-");
  listAllFilesInDir("/");

  if (LittleFS.mkdir("/a")) {
    Serial.println("ok");
  } else {
    Serial.println("ko");
  }

  Serial.println("-2-");
  listAllFilesInDir("/");

  if (LittleFS.mkdir("/a/b")) {
    Serial.println("ok");
  } else {
    Serial.println("ko");
  }

  Serial.println("-3-");
  listAllFilesInDir("/");

  if (LittleFS.rmdir("/a/b")) {
    Serial.println("ok");
  } else {
    Serial.println("ko");
  }

  Serial.println("-4-");
  listAllFilesInDir("/");
}

void loop() {
}
```
And the output it gives:

```
-1-
ok
-2-
Dir: /a/
ok
-3-
Dir: /a/
Dir: /a/b/
ok
-4-
```

earlephilhower, 2024-08-24:
Thanks for the detailed description and MCVE!  This is actually working as designed.

It's a backwards-compatibility thing because directories were not supported in SPIFFS (the first `File`).  Directories are created automatically as a file is opened (i.e. on a blank FS you can `LiffleFS.open("/this/file/is/deep/in/dirs.txt, "w")` and it will automatically created the base dirs to make it work.  When all files are `remove`d from a folder the directory is removed to avoid a bunch of leftover folders.  That's what you're running into.

So, there's no real need to `mkdir` or `rmdir`, and you don't need to worry about a directory existing before writing a file into it.

jlbirccyn, 2024-08-24:
Thanks for your quick answer Earle.

If it works as expected, it is ok :-)

Best regards

closed 2024-08-24 by jlbirccyn