earlephilhower/arduino-pico #2359: Feature Request: Change from ARM (M33) to RISC-V (Hazard3)

vslinuxdotnet, 2024-08-26:
Hello,
Is possible to add a option menu flag, to change from ARM (M33) to  RISC-V (Hazard3) ?

EX:
cmake -DPICO_PLATFORM=rp2350-arm-s -DPICO_BOARD=pico2 ..
cmake -DPICO_PLATFORM=rp2350-riscv -DPICO_BOARD=pico2 ..

Thanks,
VS

vslinuxdotnet, 2024-08-26:
Seams that will require also the riscv-gnu-toolchain

earlephilhower, 2024-08-26:
Yes, a new toolchain is needed to be added and built in [pico-quick-toolchain](https://github.com/earlephilhower/pico-quick-toolchain) (we have special Newlib configurations to fully support multicore and FreeRTOS, which is not present int any generic build).  What's worse, though, is it seems only bleeding-edge GCC 14.1 supports the full Hazard3 ISA, which means a bump to GCC revs...always lots of little gotchas in that.

Then the core itself will need revision to support the new HW in terms of IRQs, DMA, GPIO control.  Some may "just work" but for parts not covered by the SDK there will be porting involved...

Hedda, 2024-08-27:
Edge case but maybe also consider running mixed mode with one RISC-V core and one ARM Cortex-M33 core which the RP2350 supports? Arm/RISC-V switching is explained in the official datasheet: as "_They're selectable at boot time: Each port into the bus fabric can be connected either to an M33 or a Hazard3 via a mux. **You can even, if you're feeling obtuse, run with one of each.**_"

earlephilhower, 2024-09-23:
See #2491 .  Not sure if it will really be part of the core...

closed 2024-09-29 by earlephilhower