earlephilhower/arduino-pico #2384: Tool Manager: Installing git+https://github.com/earlephilhower/arduino-pico.git installation failed

nth1314, 2024-08-29:
Tool Manager: Installing git+https://github.com/earlephilhower/arduino-pico.git
git version 2.46.0.windows.1
Cloning into 'C:\Users\Administrator\.platformio\.cache\tmp\pkg-installing-z2ytds_k'...
fatal: unable to access 'https://github.com/earlephilhower/arduino-pico.git/': Failed to connect to github.com port 443 after 21011 ms: Could not connect to server
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\Administrator\\.platformio\\.cache\\tmp\\pkg-installing-z2ytds_k']

earlephilhower, 2024-08-29:
> Failed to connect to github.com port 443 after 21011 ms: Could not connect to server

It says exacting what the problem is right there.   Your computer couldn't connect to GitHub.  Either it was down when you ran this, or your network had problems. You should try again and if that fails you'll need to debug your network issue.

closed 2024-08-29 by earlephilhower

nth1314, 2024-08-29:
Thank you very much for your reply, my browser can open github perfectly, I will check my network again

> It says exacting what the problem is right there. Your computer couldn't connect to GitHub. Either it was down when you ran this, or your network had problems. You should try again and if that fails [...]

Thank you very much for your reply, my browser can open github perfectly, I will check my network again