earlephilhower/arduino-pico #2389: How to set up LittleFS for usage with the Updater library

LordOfDeadbush, 2024-08-30:
Hello,
I have been having some issues understanding the documentation at [https://arduino-pico.readthedocs.io/en/latest/ota.html](the OTA section) of the documentation. 
Is there an example or a specific set of steps required to initialize the LittleFS object? I have been observing hanging on the read from stream method. 
I noticed a general lack of documentation on this library on both this and the ESP8266 documentation. Is there a place where this is better documented?
Corn

earlephilhower, 2024-08-30:
Every filesystem example shows initialization, just `LittleFS.begin()`.  Make sure you have a filesystem set in the menus, of course

closed 2024-08-30 by earlephilhower

LordOfDeadbush, 2024-08-30:
Thank you Earle!
A quick follow up question - I seem to be hanging on LittleFS.begin() now, is there any reason that could be happening? I've tried moving it to different parts of the setup() to no avail. I am using Arduino IDE with 1mb FS. 
Is it possible I am missing a crucial part of setup?
Corn

earlephilhower, 2024-08-30:
Does the Speedtest.ino example run?  That's a very basic filesystem test.  There really isn't anything you need to do to start up the filesystem.  You can also force a format, and that's shown in the example above I believe.

LordOfDeadbush, 2024-08-30:
Hello,
I tried your advice. Speedtest.ino runs properly, but for some reason it still hangs in my code, even when I force formatting of the filesystem. 
Is there a way to log debug messages from Update to the console to see what it doesn't like?
Corn

LordOfDeadbush, 2024-08-30:
to clarify, Speedtest.ino works properly, but when I use LittleFS.begin() in my code it still hangs

LordOfDeadbush, 2024-08-30:
ok so update:
for some reason it works now, changed the position in my code and now it works???
Update is still hanging at `Update.begin()`

earlephilhower, 2024-08-30:
Without a small MCVE there's not really much I can suggest other than looking into the source code of `libraries/ArduinoOTA` or the `PicoOTA` examples (since it just needs a file on the FS to do an update, using the `Updater` class isn't really needed....see the OTAFromFile example)

LordOfDeadbush, 2024-09-03:
Hey Earle,
Did some digging through my code to find the culprit, seems to be an issue with multicore stuff.

The following code hangs:
```
#define LED_PIN 25 // this is standard on most picos
#include "LittleFS.h"
#include <FS.h>
File file;

void setup() {
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  rp2040.idleOtherCore(); // make core 1 wait for core 0 to setup

  // put your setup code here, to run once:
  gpio_put(LED_PIN, 1);
  LittleFS.format();
  gpio_put(LED_PIN, 0);
  bool fs_status = LittleFS.begin();
  file = LittleFS.open("file.bin", "w+");

}

void loop() {
  // put your main code here, to run repeatedly:

}

void setup1() {
  // ...
}

void loop1() {
  // ...
}
```

And this code works (LittleFS stuff happens before the core1 idle)
```
#define LED_PIN 25 // this is standard on most picos
#include "LittleFS.h"
#include <FS.h>
File file;

void setup() {
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  // put your setup code here, to run once:
  gpio_put(LED_PIN, 1);
  LittleFS.format();
  gpio_put(LED_PIN, 0);
  bool fs_status = LittleFS.begin();
  file = LittleFS.open("file.bin", "w+");

  rp2040.idleOtherCore(); // make core 1 wait for core 0 to setup
}

void loop() {
  // put your main code here, to run repeatedly:

}

void setup1() {
  // ...
}

void loop1() {
  // ...
}
```

Any idea what could be going on here? I think a possibility could be the internal code of LittleFS is interacting with `idleOtherCore()` in a weird way, which could be causing this hang.
Will continue investigating this on my end, do you have any idea how I could get around this?
Corn

LordOfDeadbush, 2024-09-03:
Note: this is on the W5500-evb-pico in case I didn't mention this earlier in the thread

LordOfDeadbush, 2024-09-03:
Through some further testing I've noticed that this also won't work:

```
#define LED_PIN 25 // this is standard on most picos
#include "LittleFS.h"
#include <FS.h>
File file;

void setup() {
  Serial.begin(115200);
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  // put your setup code here, to run once:

  rp2040.idleOtherCore(); // make core 1 wait for core 0 to setup
  sleep_ms(20000);
  rp2040.restartCore1();
  gpio_put(LED_PIN, 1);
  LittleFS.format();
  gpio_put(LED_PIN, 0);
  bool fs_status = LittleFS.begin();
  file = LittleFS.open("file.bin", "w+");
  Serial.println("done");
}

void loop() {
  // put your main code here, to run repeatedly:

}

void setup1() {

}

void loop1() {

}
```

but when I do it with `resumeOtherCore` it works for some reason:
```
#define LED_PIN 25 // this is standard on most picos
#include "LittleFS.h"
#include <FS.h>
File file;

void setup() {
  Serial.begin(115200);
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  // put your setup code here, to run once:

  rp2040.idleOtherCore(); // make core 1 wait for core 0 to setup
  sleep_ms(20000);
  rp2040.resumeOtherCore();
  gpio_put(LED_PIN, 1);
  LittleFS.format();
  gpio_put(LED_PIN, 0);
  bool fs_status = LittleFS.begin();
  file = LittleFS.open("file.bin", "w+");
  Serial.println("done");
}

void loop() {
  // put your main code here, to run repeatedly:

}

void setup1() {

}

void loop1() {

}
```
is there a reason why these could be working differently?
Corn

earlephilhower, 2024-09-03:
Do not run `idleOtherCore` and expect things to work.

Remove all instances of that and try whatever you were doing again.  If you need to block another core, use something like a `mutex` or shared global, not this call.  

It is only intended for naked flash erase or write access.  It is not a general purpose "pause" call.

LordOfDeadbush, 2024-09-03:
Thank you so much Earle! this seems to work great!!!
Corn