earlephilhower/arduino-pico #2396: Can program Pico with 3.9.5 but not 4.0.1

technoblogy, 2024-08-31:
With 4.0.1 I get the error:
````text
dyld: Library not loaded: @executable_path/libusb-1.0.0.dylib
  Referenced from: /Users/david/Library/Arduino15/packages/rp2040/tools/pqt-picotool/2.2.0-8a9af99/picotool
  Reason: Incompatible library version: picotool requires version 5.0.0 or later, but libusb-1.0.0.dylib provides version 3.0.0
...
signal: abort trap
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board Raspberry Pi Pico.
````
It works on 3.9.5, but I need 4.0.1 to program a Pico 2 which gives the same error.

I'm using Arduino IDE 1.8.19 on MacOS 10.13 on an Intel Mac.

Thanks, David

earlephilhower, 2024-08-31:
I think you have an old version of libusb installed and the tool is grabbing that and not the one that's shipped.  Looking at a Mac86 CI run [here](https://github.com/earlephilhower/arduino-pico/actions/runs/10647471622/job/29515439540?pr=2400)  we see it's all good (and if Macs don't build CI is set to fail).

It's set to link to the shipped dynlib, but maybe MacOS will grab system versions first:
````
....
 picotool v2.0.0 (Darwin, AppleClang-13.0.0.13000029, Release)
./system/picotool/picotool:
	@executable_path/libusb-1.0.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
...
````

A run of `otool -L <installed-picotool-exe>` will give your specific machine report.

You can try whatever `brew` command will update your installation (`brew install libusb` ?) or uninstall libusb completely and try again.

technoblogy, 2024-08-31:
Thanks for the suggestion. I did:

    HardDisk-318:~ david$ brew uninstall --ignore-dependencies libusb
    Uninstalling /usr/local/Cellar/libusb/1.0.22... (29 files, 514.8KB)

but I still get the same error when trying to program Pico with 4.0.1.

earlephilhower, 2024-08-31:
What does `otool` report on your executable?  And can you do a `ls -l` on the picotool dir to verify it's actually in there?

There was a repackage post 4.0.1 but only to add a `package.json` for Platform.IO.  In any case the 4.0.1 `picotool` also [ran in CI](https://github.com/earlephilhower/arduino-pico/actions/runs/10553816550/job/29234701681).

Basically, it looks like there's some older version of the libusb file in your LDPATH (or whatever the Mac/BSD equivalent is) that the binary is trying to grab, not the distributed one.

technoblogy, 2024-09-01:
> What does otool -L <pth-to-picotool-eve> report on your executable?
````text
/Users/david/Library/Arduino15/packages/rp2040/tools/pqt-picotool/2.2.0-8a9af99/picotool:
	@executable_path/libusb-1.0.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)

````
> And can you do a ls -l on the picotool dir to verify it's actually in there?
````text
total 4824
-rw-r--r--  1 david  staff     1489 18 Aug 02:22 LICENSE.TXT
-rw-r--r--  1 david  staff    44266 18 Aug 02:22 README.md
-rw-r--r--  1 david  staff   159048 31 Jan  2024 libusb-1.0.0.dylib
lrwxr-xr-x  1 david  staff       18 31 Aug 14:36 libusb-1.0.dylib -> libusb-1.0.0.dylib
-rwxr-xr-x  1 david  staff  2259344 18 Aug 02:24 picotool
````

earlephilhower, 2024-09-01:
I think you have somehow ended up with old versions of libusb.dynlib and new picotool binary in the directory.

The [tarball in the pqt release](https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-apple-darwin20.4.picotool-8a9af99.240818.tar.gz) shows a data of Aug 11, 2024, not Jan 31 2024.

Can you remove the directory completely (rm -r), remove the Arduino-Pico install using the Boards Manager, and then reinstall?

technoblogy, 2024-09-01:
Thank you for trying to help on this. I've uninstalled and reinstalled as suggested, and still get the same file dates:
````text
total 4824
-rw-r--r--  1 david  staff     1489 18 Aug 02:22 LICENSE.TXT
-rw-r--r--  1 david  staff    44266 18 Aug 02:22 README.md
-rw-r--r--  1 david  staff   159048 31 Jan  2024 libusb-1.0.0.dylib
lrwxr-xr-x  1 david  staff       18  1 Sep 18:29 libusb-1.0.dylib -> libusb-1.0.0.dylib
-rwxr-xr-x  1 david  staff  2259344 18 Aug 02:24 picotool
````
I've also confirmed that uploading to the Pico or Pico2 still fails in the same way.

earlephilhower, 2024-09-01:
Whelp, you have a haunted Mac.  CI and another Mac are running fine, so there's something particularly hosed on this specific one.

We used to perform exorcisms on the 68030 based NuBus Macs I wrote device drivers for in college, but I think those went out of style around the PowerPC transition.  Didn't run on the Rosetta emulator and all...

Anyway, I suggest just ignoring the issue and overwriting the files with the ones from the release.  Download https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-apple-darwin20.4.picotool-8a9af99.240818.tar.gz and open it up in whatever archive manager Macs have today (or use `tar xvf file.tar.gz` from a command prompt) and manually copy the proper dynlib into the picotool install directory (maybe the picotool exe as well, who knows WTH is there now).

technoblogy, 2024-09-01:
OK, thanks, now I've got:
````text
total 4824
-rw-r--r--@ 1 david  staff     1489 18 Aug 02:22 LICENSE.TXT
-rw-r--r--@ 1 david  staff    44266 18 Aug 02:22 README.md
-rw-r--r--@ 1 david  staff   159048 11 Aug 22:37 libusb-1.0.0.dylib
lrwxr-xr-x@ 1 david  staff       18 31 Jan  2024 libusb-1.0.dylib -> libusb-1.0.0.dylib
-rwxr-xr-x@ 1 david  staff  2259344 18 Aug 02:24 picotool
````
but when I try and upload I still get the error:
````text
dyld: Library not loaded: @executable_path/libusb-1.0.0.dylib
  Referenced from: /Users/david/Library/Arduino15/packages/rp2040/tools/pqt-picotool/2.2.0-8a9af99/picotool
  Reason: Incompatible library version: picotool requires version 5.0.0 or later, but libusb-1.0.0.dylib provides version 3.0.0
...
signal: abort trap
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board Raspberry Pi Pico 2.

earlephilhower, 2024-09-01:
At this point you really have a haunted machine and there's not much I can offer you.

I suggest downloading the [x86_64-apple-darwin20.4.picotool-8a9af99.240818.tar.gz](https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-apple-darwin20.4.picotool-8a9af99.240818.tar.gz) tarball.  Then from a command prompt expand it and try running picotool manually
````
$ tar zxvf x86_64-apple-darwin20.4.picotool-8a9af99.240818.tar.gz
$ ./picotool/picotool --version
````

If that doesn't work, then your Mac itself has some kind of brain damaged fixation on some older version of the file.  Maybe it was loaded and never unloaded and the OS is trying to share it or something.  Could try rebooting and doing a `find` for other version of `libusb-1.0.0.dylib` and blowing them away.

closed 2024-09-01 by earlephilhower

earlephilhower, 2024-09-01:
I've manually checked using `docker` and the distributed dylib is 5.0.0:
````
user@4e8f679fdc82:/workdir/aaa$ !tar
tar xvf ../x86_64-apple-darwin20.4.picotool-8a9af99.240818.tar.gz
picotool/
picotool/libusb-1.0.dylib
picotool/libusb-1.0.0.dylib
picotool/README.md
picotool/LICENSE.TXT
picotool/picotool
user@4e8f679fdc82:/workdir/aaa$ cd picotool/
user@4e8f679fdc82:/workdir/aaa/picotool$ x86_64-apple-darwin20.4-otool -L libusb-1.0.dylib 
libusb-1.0.dylib:
	/usr/local/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1858.112.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 60158.100.133)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
````

technoblogy, 2024-09-02:
OK, thank you for trying. I think my version of MacOS is probably just too old.

technoblogy, 2024-09-02:
In case it's useful, I've understood a bit more about what the problem is:

I wrote:

> It works on 3.9.5, but I need 4.0.1 to program a Pico 2 which gives the same error.

The difference is that with the **Upload Method: "Default (UF2)"**, core version 3.9.5 doesn't run `picotool`, but version 4.0.1 runs `picotool` with the command `picotool uf2 convert`, even for the RP2040.

I assume that's something that's necessary for the RP2350, so it's easier to do it in every case.

earlephilhower, 2024-09-02:
Yes, what has happened is that RPI killed the `elf2uf2` binary and replaced it with a function inside `picotool`.  So it's needed even on the RP2040.  (On the RP2350 it adds bootrom headers/partitions/etc. to make the chip's ROM know what CPU to use and actually run it.)

technoblogy, 2024-10-05:
If anyone has a workaround to get `picotool` to run on MacOS 10.13 (High Sierra) or MacOS 10.14 (Mojave) I'd be very grateful, so I don't have to use a different laptop for my RP2350 development.

earlephilhower, 2024-10-05:
You can try building the SDK from sources on your machine.  There are no changes to code in the sdk tools used here.

technoblogy, 2024-10-05:
OK, I'll see if I can get that to work - thanks.