earlephilhower/arduino-pico #2397: IRQ stack overflow

Yveaux, 2024-08-31:
Hi there!
I'm working on a project which combines USB host functionality, FreeRTOS and a number of interrupt sources.
Every now and then the system panics with "IRQ stack overflow" which traces back to `wiring_private.cpp`.

The standard `maxIRQs` appears to be too low for my application: https://github.com/earlephilhower/arduino-pico/blob/226a3188975b21d29d32fe4423a6c929766192a5/cores/rp2040/wiring_private.cpp#L28

Raising it to 30 makes the panics disappear.

Would it be possible to define the size of the stack as a global variable?

earlephilhower, 2024-08-31:
That's a pretty massive amount of nested `noInterrupt()`s!  Are you sure you're not over-disabling things?

The stack needs to be setup during global object instantiation, even before `main` runs.  It can't be a variable, since you can't make a global array with a variable size, but I suppose we can let it be overridden it with a `build.local` file, so it'd be
````
#ifndef maxIRQs
#define maxIRQs 15
#endif
````

Changing defaults it in the core would be a non-starter because that would be a non-trivial amount of RAM being eaten (and unused) for all other apps.

Want to make a PR implementing that `ifndef` block?

Yveaux, 2024-09-01:
> That's a pretty massive amount of nested noInterrupt()s! Are you sure you're not over-disabling things?

It's been a while since I looked into this issue, but according to my notes the cause seemed to lie in the fact that FreeRTOS task preemptions are not blocked when calling `noInterrupts()`.

> Want to make a PR implementing that `ifndef` block?

Sure, I created #2401

earlephilhower, 2024-09-01:
Closed by #2401

closed 2024-09-01 by earlephilhower