earlephilhower/arduino-pico #2398: Reconfigure FreeRTOS

Yveaux, 2024-08-31:
Hi there!

Would it be possible to use your own `FreeRTOSConfig.h` file?

The one included in the library (`libraries/FreeRTOS/src/FreeRTOSConfig.h`) e.g. doesn't have configUSE_APPLICATION_TASK_TAG set which I use to debug task switches.

I found a way around this by patching the `FreeRTOSConfig.h` included in arduino-pico, but figured there could be a cleaner solution to this problem.
I see 2 possible solutions: 
- allow inclusion of one's own `FreeRTOSConfig.h`
- allow overwriting any of the defines in a global configuration, by wrapping them in `#ifndef`s, like

> #ifndef configUSE_APPLICATION_TASK_TAG
>  #define configUSE_APPLICATION_TASK_TAG    0
> #endif

What's your thought on this?

earlephilhower, 2024-09-01:
I'm far from a FreeRTOS expert (although I really dug into the internals when I was breaking and un-breaking it for the RP2040 + flash write + Newlib reeent support!) so I can't think of a clean way of the 1st option.  Can you suggest something?  Maybe using the GCC-only `__has_include` to make it optional and not break existing code.  OTW, if you'd like to wrap every `define` with an `ifndef` in the config file that's fine by me, too.

Yveaux, 2024-09-01:
I'm unfamiliar with `__has_include` but I'm going to have a look (always nice to learn something new ;-)
I'll prepare a PR.

Yveaux, 2024-09-02:
Please have a look at #2407
Using `__has_include` would work, but it would also require anyone using it to exactly clone the complete `FreeRTOSConfig.h` file in one's project. If the original config is updated (e.g. due to an update of arduino-pico) things may break.
IMO a cleaner option is to individually override config items, for which an implementation you can find in the PR.

commit 48c4c202 (2024-09-03): Allow reconfiguring individual FreeRTOS config items - fixes #2398 (#2407)
closed 2024-09-03 by earlephilhower