earlephilhower/arduino-pico #2399: Error compiling USB Host only on FreeRTOS

Yveaux, 2024-08-31:
When compiling USB Host only under FreeRTOS (`CFG_TUH_ENABLED` = 1, `CFG_TUD_ENABLED` = 0) a compile error is raised:

```
C:\Users\...\.platformio\packages\framework-arduinopico\libraries\FreeRTOS\src\variantHooks.cpp: In function 'void __usb(void*)':
C:\Users\...\.platformio\packages\framework-arduinopico\libraries\FreeRTOS\src\variantHooks.cpp:494:13: error: 'tud_task' was not declared in this scope; did you mean 'tuh_task'?
  494 |             tud_task();
      |             ^~~~~~~~
      |             tuh_task
```
As `tud_task()` is only implemented for USB devices, not for hosts.

A simple fix is only calling the semaphore code and `tud_task()` when CFG_TUD_ENABLED:
```
while (true) {
#if CFG_TUD_ENABLED
        auto m = __get_freertos_mutex_for_ptr(&__usb_mutex);
        if (xSemaphoreTake(m, 0)) {
            tud_task();
            xSemaphoreGive(m);
        }
#endif
        vTaskDelay(1 / portTICK_PERIOD_MS);
}
```
or even let the `__usb()` task return.

earlephilhower, 2024-08-31:
What USB menu config are you building with, exactly?  All 4 menu entries build the following FreeRTOS test fine:
````
#include <FreeRTOS.h>
void setup() {
}
void loop() {
}
````
I tried SDK, TinyUSB, TinyUSB Host, and No USB without error.

Yveaux, 2024-09-01:
Sorry, but the setup is over half a year old and more complex than I thought initially: Adafruit_TinyUSB has been disabled, and the (then) latest TinyUSB was pulled into the project, all compiled under platform.io.

This is all working (except for the mentioned issue) but distilling a simple standalone example to show the issue is less trivial.

Let's ignore this for now; I may come back to it at a later time.

Thanks for your effort!

closed 2024-09-01 by Yveaux