earlephilhower/arduino-pico #2403: FSInfo64 and LittleFS.info64() not available

jlbirccyn, 2024-09-01:
Hello,

Trying to use ```FSInfo64``` and ```LittleFS.info64()``` according to the documentation that says ```FSInfo``` and ```LittelFS.info()``` are deprecated, I got a compilation error:

```
/Users/jlb/Documents/Arduino/WiFiServer-test/Commands.cpp: In function 'void infoCommandFunction(CommandInterpreter&, byte)':
/Users/jlb/Documents/Arduino/WiFiServer-test/Commands.cpp:329:3: error: 'FSInfo64' was not declared in this scope; did you mean 'FSInfo'?
  329 |   FSInfo64 fsinfo;
      |   ^~~~~~~~
      |   FSInfo
/Users/jlb/Documents/Arduino/WiFiServer-test/Commands.cpp:330:12: error: 'class fs::FS' has no member named 'info64'; did you mean 'info'?
  330 |   LittleFS.info64(fsinfo);
      |            ^~~~~~
      |            info
/Users/jlb/Documents/Arduino/WiFiServer-test/Commands.cpp:330:19: error: 'fsinfo' was not declared in this scope; did you mean 'FSInfo'?
  330 |   LittleFS.info64(fsinfo);
      |                   ^~~~~~
      |                   FSInfo
```

Looking at the source code, I see that the FSInfo structure has its two members ```totalBytes``` and ```usedBytes``` which are 64 bits. Have ```FSInfo64``` and ```info64()``` been removed and ```FSInfo``` updated?

Sincerely

earlephilhower, 2024-09-01:
Yes, that was part of the BREAKING release note (which I guess the IDE never is going to show).  Just move to `FSInfo` and `info` which are now 64-bit only (can you even separately buy an SD card of <4GB today??).

closed 2024-09-01 by earlephilhower

jlbirccyn, 2024-09-01:
Ok. Thanks.

 I use the second half of the flash as FS.

Would you accept a PR for an updated documentation ?

earlephilhower, 2024-09-01:
Absolutely! I *thought* I updated `fs.rst` but I guess not. :(

jlbirccyn, 2024-09-01:
I'm working on it :)

Also, further up on the same documentation page, there is:

```
LittleFS File System Limitations

The LittleFS implementation for the RP2040 supports filenames of up
to 31 characters + terminating zero (i.e. ``char filename[32]``), and
as many subdirectories as space permits.
```

If I print ```LFS_NAME_MAX``` I get 255, not 32.

earlephilhower, 2024-09-01:
Yes, that was also changed in the code but not docs.

jlbirccyn, 2024-09-01:
Ok, I change it to 255