earlephilhower/arduino-pico #2408: macOS Sequoia beta

sstaub, 2024-09-03:
With macOS Sequoia beta compiling fails. MacBook Pro 16" (2019) Intel Core i7
[image]
[image]

sstaub, 2024-09-03:
Here the output pio run --verbose

```
sstaub@9301096c-1d75-4d3f-a7c6-2fdcd238c163 arduino-blink % pio run --verbose
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board_build.core: earlephilhower; board_build.filesystem_size: 0.5m; board: rpipico; platform_packages: framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#master; build_flags: -Wall)
----------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.196d31b) (git+https://github.com/maxgerhardt/platform-raspberrypi.git) > Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40001.0+sha.6a5f98c (git+https://github.com/earlephilhower/arduino-pico.git#master) 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0, https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-apple-darwin20.4.arm-none-eabi-d04e724.240125.tar.gz)
Flash size: 2.00MB
Sketch size: 1.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 1568768 EEPROM start: 0x101ff000 Filesystem start: 0x1017f000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Ignored library /Users/sstaub/.platformio/packages/framework-arduinopico/libraries/Adafruit_TinyUSB_Arduino
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 64 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio/build/pico/firmware.elf -T "/Users/sstaub/Dateien/Development/Raspberry Pico/arduino-blink/.pio/build/pico/memmap_default.ld" @/Users/sstaub/.platformio/packages/framework-arduinopico/lib/rp2040/platform_wrap.txt @/Users/sstaub/.platformio/packages/framework-arduinopico/lib/core_wrap.txt -u_printf_float -u_scanf_float -Wl,--no-warn-rwx-segments -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--undefined=runtime_init_install_ram_vector_table -march=armv6-m -mcpu=cortex-m0plus -mthumb .pio/build/pico/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o .pio/build/pico/src/main.cpp.o -L.pio/build/pico -Wl,--start-group .pio/build/pico/libFrameworkArduino.a /Users/sstaub/.platformio/packages/framework-arduinopico/lib/rp2040/ota.o /Users/sstaub/.platformio/packages/framework-arduinopico/lib/rp2040/libpico.a /Users/sstaub/.platformio/packages/framework-arduinopico/lib/rp2040/libipv4.a /Users/sstaub/.platformio/packages/framework-arduinopico/lib/rp2040/libbearssl.a -lm -lc -lstdc++ -lc -Wl,--end-group
/Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: section .stack1_dummy VMA [20040000,200407ff] overlaps section .heap VMA [200024c4,20041fff]
collect2: error: ld returned 1 exit status
*** [.pio/build/pico/firmware.elf] Error 1
=================================================== [FAILED] Took 0.89 seconds ===================================================
sstaub@9301096c-1d75-4d3f-a7c6-2fdcd238c163 arduino-blink % ```

earlephilhower, 2024-09-03:
I think you're running P.IO before it's been updated to 4.0.1 JSONs.  That log doesn't match the latest output from our own CI:
````
Processing rpipico (platform: raspberrypi; board: rpipico; framework: arduino; platform_packages: framework-arduinopico@symlink:///home/runner/work/arduino-pico/arduino-pico)
--------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.196d31b) (git+https://github.com/maxgerhardt/platform-raspberrypi.git) > Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
....
````

Note the `HARDWARE` line shows 264K and not 256K like is corrected in the current files  (it was 256K+8K before because it listed the private stacks as RAM which is technically correct but not really useful since only 256K is there for DATA/HEAP).

sstaub, 2024-09-03:
Meanwhile I deleted all PIO files and reinstalled PIO. But there is no change. 
How to change the .ini to use an older release? I found nothing about in the docs.

earlephilhower, 2024-09-03:
You can manually pull in the JSON files from `tools/json` to wherever P.IO installs them and try again.

sstaub, 2024-09-03:
I replaced the package with version 3.9.5 and it works, however.

sstaub, 2024-09-03:
There is the same problem on my MacBook Pro 13" M1 with macOS Sonoma.

earlephilhower, 2024-09-03:
Again, this has nothing to do with the OS and earlier release JSONs match the un-updated p.io script.  You can either update the JSONs as mentioned above or use 3.9.x until Max has the time to update his PIO wrapper script with the new JSONs (and the 2350 won't work either until the same is done), of course.

maxgerhardt, 2024-09-03:
```cpp
: section .stack1_dummy VMA [20040000,200407ff] overlaps section .heap VMA [200024c4,20041fff]
```
This either means you sketch used so much statically allocated objects that the stack overflowed into the heap area, or there's something wrong with the linker script that checks for that collision.

If you want to use the 4.x Arduino-Pico core, then `Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; ` is wrong. It must be `https://github.com/maxgerhardt/platform-raspberrypi.git#rp2350_picosdk`.

Let me take a look at this in peace to come up with the rest of the answer of why it doesn't link...

earlephilhower, 2024-09-03:
I think that's due to the heap mods done in the Pico SDK 2.0.  Before, the heap was set small and expanded using `sbrk` as needed up to the stack start.  Now, they make it eat the whole area (I don't understand their reasoning, but they say they had issues w/the sbek/heap expansion in certain configurations).

In 4.0 I also got rid of the weird logic implemented to override the JSON RAM size in platform-build.py.  So the new JSONs have 256K, not 264K, and no longer need that logic (makes the RP2350 implementation cleaner and removed one weird unneeded bit of code).  BUT if you use the NEW platform-build.py and the OLD JSONs, I think the heap end will be set to 264K, not 256K, hence the overlap error shown by `ld`.

Architeuthis-Flux, 2024-09-03:
> You can manually pull in the JSON files from `tools/json` to wherever P.IO installs them and try again.

Came here with the same issue, and that fixed it. Thanks!

For anyone else who has this problem, the board definition JSON files should be in 

`/Users/{USER}/.platformio/packages/framework-arduinopico/tools/json/`

Then I just copied rpipico.json to my PlatformIO project in the `{PROJECT_NAME}/boards/` folder.

[image]

earlephilhower, 2024-09-03:
Actually, I think you can use `https://github.com/maxgerhardt/platform-raspberrypi.git#rp2350_picosdk` PIO script and not copy anything. I'm not sure why there's a separate branch, but hopefully it can be merged back to the main one there soon.  There doesn't seem to be any major breakage w/the 4.0 release...

maxgerhardt, 2024-09-03:
That `rp2350_picosdk` branch will be merged back to `main` and then deleted shortly. The CI is just complaining now that OpenSSL can't be installed for the signed OTA tests.. the regular builds work fine.

maxgerhardt, 2024-09-03:
https://github.com/maxgerhardt/platform-raspberrypi/pull/72 is merged into mainline.

I've verified a blinky on Pico, Pico W and my Pimeroni RP2350 to be working, along with debugging.

Delete `~/.platformio/rasperrypi` as needed to trigger the update upon the next build.

closed 2024-09-03 by earlephilhower

sstaub, 2024-09-03:
@earlephilhower @maxgerhardt  Thank you for the excellent support.